libremesh/lime-packages #953: Move uhttpd.main.redirect_https setting from lime-system to shared-state and ubus-lime-metrics

ilario, 2022-11-30:
As spotted by @hiurequeiroz, when using lime-system without uhttpd, an error appears when running lime-config:

```
Applying generic configs:
 uhttpd_https
 Error on generic config uci_set: uhttpd.main.redirect_https=0
```

This has been implemented in #799 and the error should not have any impact, but for avoiding the error message we can remove these lines from lime-system:
https://github.com/libremesh/lime-packages/blob/78162b4d773c9e8d01f63801ce3464ea84ac89b6/packages/lime-system/files/etc/config/lime-defaults#L77-L78

and add them to an uci-defaults script inside the packages which benefit from this option (the ones which frequently fetch stuff via http/https) which are shared-state and ubus-lime-metrics.

If we want to avoid using uci-defaults scripts (as suggested by @G10h4ck here: https://github.com/libremesh/lime-packages/issues/927#issuecomment-1329384326) we would need something new like a `/etc/config/lime-defaults.d/` directory (is this worth it???).

config generic_uci_config uhttpd_https
        list uci_set 'uhttpd.main.redirect_https=1'

G10h4ck, 2022-12-01:
> If we want to avoid using uci-defaults scripts (as suggested by @G10h4ck here: [#927 (comment)](https://github.com/libremesh/lime-packages/issues/927#issuecomment-1329384326)) we would need something new like a `/etc/config/lime-defaults.d/` directory (is this worth it???).

It is not useful to substitute `/etc/uci-defaults` with `etc/config/lime-defaults.d/` what I wanted to push is that we should do as much as possible at compile time, and not at runtime first-boot like `/etc/uci-defaults` but if something need to be done at runtime at first-boot `/etc/uci-defaults` is still the best place.

ilario, 2024-11-07:
Two years without alternative proposals, so uci-defaults should be ok.
Anyway this issue has a low priority.