libremesh/lime-packages #957: lime-app: received "Cannot use Radio 0, it's being used for mesh links" when "Connect to a Mobile Hotspot"

ilario, 2022-12-08:
While testing #950 on OpenWrt 22.03 I installed lime-app on OpenWrt (it pulled as dependencies most of LibreMesh packages) and tried to activate the client function to connect to a mobile hotspot.
I got the following error:
```
Cannot use Radio 0, it's being used for mesh links
```

[image]

So I imagine this works with the default configuration from LibreRouterOS ( https://gitlab.com/librerouter/librerouteros/-/blob/librerouter-1.5/files/etc/config/lime-defaults ) but not with the official default configuration ( https://github.com/libremesh/lime-packages/blob/master/packages/lime-system/files/etc/config/lime-defaults )??????????????????????????????????????????????????

The error message should at least suggest a solution.

ilario, 2023-03-20:
@selankon @fnbellomo @luandro @spiccinini can anyone look into this?

ilario, 2023-04-07:
This error comes from here:

https://github.com/libremesh/lime-app/blob/5f7c77407012c351c23b8a5e0b0510efde00b557/plugins/lime-plugin-node-admin/src/screens/hotspot/src/hotspotPage.js#L95-L98

That comes from here:

https://github.com/libremesh/lime-packages/blob/1122385d95452e631fe36ab39ab079a2725b24ab/packages/ubus-lime-utils/files/usr/lib/lua/lime/hotspot_wwan.lua#L67-L74

sanekto, 2023-04-07:
I believe the main idea is to use a radio that is not currently being used in the mesh by the device, so you don't remove the device from the mesh when using this feature, because configuring an interface as a client will break the mesh connection (the channel will be different most of the times). So devices with only one radio, or with all its radios used in the mesh are not allowed to connect to a mobile hotspot.
Yes indeed, a better message should be shown! Something like "this devices has no hotspot support ...."

spiccinini, 2023-04-07:
I believe the main idea is to use a radio that is not currently being used in the mesh by the device, so you don't remove the device from the mesh when using this feature, because configuring an interface as a client will break the mesh connection (the channel will be different most of the times). So devices with only one radio, or with all its radios used in the mesh are not allowed to connect to a mobile hotspot.
Yes indeed, a better message should be shown! Something like "this devices has no hotspot support ...."

ilario, 2023-04-07:
Yep, the error message should surely point the user to how to fix, if possible. Something like "do this and that for removing mesh on the 2.4 GHz interface. And then try again."

ilario, 2023-04-07:
> or with all its radios used in the mesh are not allowed to connect to a mobile hotspot.

I think that the problem arises by the fact that all the LibreMesh devices, by default, use all radios for mesh. So this is a problem for every LibreMesh user, unless they configured their router for not using radio0 for mesh.

ilario, 2024-11-07:
> all the LibreMesh devices, by default, use all radios for mesh. So this is a problem for every LibreMesh user, unless they configured their router for not using radio0 for mesh.

So this is just another feature designed for working with LibreRouterOS regardless of its usage with LibreMesh.
Is there a way for hiding this feature from lime-app? @selankon 
Does lime-app accept a list of tabs to disable?
Or is there a way to disable it at compilation time saving also some precious kB?