libremesh/lime-packages #960: Rpcd SIGSEGV when wrong password

selankon, 2022-12-20:
I observed a strange behavior when trying to do login with wrong password. 

## How to reproduce

I'm using a TP-Link n600 with LibrerouterOs and lime-packages from c9aa653a and lime-app on https://github.com/libremesh/lime-app/commit/53c44898d1f5ad18be0083153afab87a781fc848 .

When I start the lime-app and access to any page that need shared password login I observed that it try first a login with [this](https://github.com/libremesh/lime-app/blob/53c44898d1f5ad18be0083153afab87a781fc848/src/containers/SharedPasswordLogin/sharedPasswordLogin.js#L67):

```
login({username: 'root', password: ''});
```

This causes a 5 seconds timout (which mean that the connection is open during 5 seconds until fail, instead of receive a wrong password answer). 

On router side, with `logread`, I can see that any petition to `session/access` with a wrong password causes an error like this:

```
Thu Mar 17 21:03:09 2022 kern.info kernel: [  433.882719] do_page_fault(): sending SIGSEGV to rpcd for invalid read access from 00000014
Thu Mar 17 21:03:09 2022 kern.info kernel: [  433.891199] epc = 0040283d in rpcd[400000+8000]
Thu Mar 17 21:03:09 2022 kern.info kernel: [  433.895923] ra  = 00402839 in rpcd[400000+8000]
Thu Mar 17 21:03:09 2022 daemon.info procd: Instance rpcd::instance1 s in a crash loop 6 crashes, 69 seconds since last crash 
```

So the timeout is because it crashes when wrong password, and shouldn't be the correct behavior (?). 

On other side, if I try to login repetitively with a wrong password, I observed that suddenly, the ubus return this on any `session/login|access` call, causing limeapp stop working: 

```
Object { code: -32000, message: "Object not found" }
```

I had to restart the router every time this happen. Lime app calls `session/login|access` when accessing landing page. I have tot test it in another hardware

selankon, 2022-12-22:
I confirm this happen also on a librerouter. 

### Update

It return a session id although invalid password is sent:

```json
{
  "ubus_rpc_session": "22c8d7bc966cadc185b20bdff77ddfef",
  "timeout": 5000,
  "expires": 4999,
//...
  "data": {
    "username": "lime-app"
  }
}
```

Its strange because on the qemu virtualization the error doesn't happen router side but it return an strange json with this value:

```json
{
  "jsonrpc": "2.0",
  "id": 16,
  "result": [
    6
  ]
}
```

ilario, 2023-02-23:
@selankon is this still happening?

selankon, 2024-01-09:
I forgot this, I'll check asap

ilario, 2024-11-07:
@selankon updates?
Does this affect the release candidate 2024.1-rc1?