libremesh/lime-packages #962: Lime utils broken if associated ap's

selankon, 2022-12-26:
If associated AP's the command:

```bash
ubus call lime-utils get_node_status '{}'
```

Is broken. Seem to happen after https://github.com/libremesh/lime-packages/pull/939

selankon, 2022-12-26:
Could be related, but some lime app views that are using redux instead of ReactQuery are broken:

- rxPage (BROKEN)
- Notes (BROKEN)
- Lim-plugin-metrics (?)
- locate (✅)
- ground-routing (?)
- Change-node (BROKEN)

Could be related to this?

ilario, 2022-12-26:
Can you provide more detailed instructions for reproducing this?

selankon, 2022-12-26:
This pr should fix it: https://github.com/libremesh/lime-packages/pull/963

Just use last master commit and check that some lime-utils ubus funtions fail
 https://github.com/libremesh/lime-packages/commit/c9aa653a67a5421aae2cb217bdbaa058f9a4cf2f

Some errors where added after merging https://github.com/libremesh/lime-packages/pull/939

closed 2023-01-18 by ilario