libremesh/lime-packages #969: shared-state-publish_dnsmasq_leases errors

pony1k, 2023-01-16:
When I run `shared-state-publish_dnsmasq_leases` it errors with:
```
/usr/bin/lua: /usr/bin/shared-state-publish_dnsmasq_leases:48: table index is nil
stack traceback:
        /usr/bin/shared-state-publish_dnsmasq_leases:48: in function 'add_dhcp_leases_from_files'
        /usr/bin/shared-state-publish_dnsmasq_leases:130: in main chunk
        [C]: ?
```
It is the following lines:
```
_, client_mac, client_ip, client_hostname, client_id = unpack(line:split(" "))
leasesTable[client_ip] = { hostname=client_hostname, mac=client_mac }
```
which have trouble because of this line in the `/tmp/dhcp.leases`-file:
```
duid 00:01:xx:xx:xx:xx:xx:xx:xx:xx:xx:xx:xx:xx`
```
I did a quick research. According to [here](https://dnsmasq-discuss.thekelleys.org.narkive.com/ia9YNLmE/dnsmasq-leases-file-format-specification), this is normal when DHCPv6 is served.

ilario, 2023-01-24:
Can you post a full `/tmp/dhcp.leases` file content (censored, if you prefer) so that we can better understand which lines should be used and which ignored?
@G10h4ck did you ever see that line?

pony1k, 2023-01-26:
Sure.
```
1674755797 28:d2:44:d8:3a:3b 10.51.147.44 katze 01:28:d2:44:d8:3a:3b
duid 00:01:00:01:2b:65:6d:6e:70:62:b8:4e:88:90
1674755842 1276192316 fd70:d3e4:df63:184d::ccbf katze 00:04:eb:cf:8f:90:ce:2c:e3:20:12:55:c6:fe:9b:c7:16:ee
1674755842 1276192316 fd43:1508:3393:fe00::ccbf katze 00:04:eb:cf:8f:90:ce:2c:e3:20:12:55:c6:fe:9b:c7:16:ee
```
Everything before the duid line is ipv4. Everything after it is ipv6. I could not find any documentation for the format.

ilario, 2023-01-26:
Looking at how the code is written, seems that is written for handling only the IPv4 case:

https://github.com/libremesh/lime-packages/blob/d632accffbb39ffb7be4444387f247f755c826f7/packages/shared-state-dnsmasq_leases/files/usr/bin/shared-state-publish_dnsmasq_leases#L47-L48

In absence of the duid error, this code would have generated broken entries for IPv6.
Did you configure some IPv6 stuff, or does your network have some IPv6 stuff that maybe are not present in most people's networks?

Anyway that code has to be adapted.
One interesting thing is also that in the IPv6 lines there is no MAC address.
So we can populate the hostTable here:
https://github.com/libremesh/lime-packages/blob/d632accffbb39ffb7be4444387f247f755c826f7/packages/shared-state-dnsmasq_leases/files/usr/bin/shared-state-publish_dnsmasq_leases#L51

but not the leasesTable in the previous listing.

pony1k, 2023-01-26:
> Did you configure some IPv6 stuff, or does your network have some IPv6 stuff that maybe are not present in most people's networks?

I have this section in my `/etc/config/dhcp`:
```
config dhcp 'lan'
        option interface 'lan'
        option start '2'
        option limit '253'
        option leasetime '1h'
        option dhcpv4 'server'
        option dhcpv6 'server'
        option ra 'server'
        option ra_slaac '1'
        list ra_flags 'managed-config'
        list ra_flags 'other-config'
        option ignore '0'
```
I think this is what one gets by default when `dnsmasq-dhcpv6` is installed and `lime-proto-anygw` is not installed. `lime-proto-anygw` would set `ignore '1'` for this section and serve only dhcpv4 on the anygw interface.

G10h4ck, 2023-01-27:
AFAIR `lime-proto-anygw` set `dnsmasq` in `ra-names` mode, so it announce the IPv6 prefix to the clients and they take an IPv6 address based on their MAC address, if they also asked an IPv4 address via DHCP, then `dnsmasq` is able to resolve their name  both to their IPv4 and IPv6 address.
Because `ra-name` is based on stateless address assignation, so is the client that take the address for itself not `dnsmasq` leasing an address the client IPv6 address doesn't end up in the lease file.

So @pony1k your setup is not supported, if it is simple to fix without breaking the `lime-proto-anygw`  use case and with some comments in the code explaining it, I am ok to review and merge a pull request that make `shared-state-publish_dnsmasq_leases` works in your setup too.

Cheers

closed 2023-01-27 by G10h4ck

ilario, 2023-01-27:
@G10h4ck I heard more than once people not including lime-proto-anygw in their firmwares, so in my opinion we should try to have things not explode in its absence.
If I recall it correctly, @digitigrafo and Gothos are not using it.

Please check the code in #975 as this issue seems simple enough to fix.

PR #975 (2023-02-15): shared-state-publish_dnsmasq_leases recognize IPv6 when IPv6 leases are present
Fix #969
I still have to test the modified code.
@pony1k can you test this?