libremesh/lime-packages #993: How to prioritize wired links？

ggsnpdsn, 2023-03-29:
I have four nodes running libremesh, connected one by one through a wired network, the first connected to the internet, and the fifth node moving around these four nodes. I find that I always access the internet through the path directly connected to the first node. I expect the fifth node to select the node with the strongest signal to access the internet. Do I need to set the batman adv parameter? How to change it?

ilario, 2023-03-30:
Hi @ggsnpdsn !
How do you see that you "always access the internet through the path directly connected to the first node"?
I don't know if there is a smart way to have the routing protocols using the cable over the wifi. I hope they will do it when they see that the packets loss on the wifi is higher than the one on the cable...

ilario, 2023-04-07:
Try removing this line:

https://github.com/libremesh/lime-packages/blob/7664098971131f486184818de8b82732fb402b87/packages/lime-proto-babeld/files/usr/lib/lua/lime/proto/babeld.lua#L127

ilario, 2023-04-13:
@ggsnpdsn please let us know if the suggestion helps!
In the positive case, I think that we can replace that line with an option that allows the user to have it or not, but leaving it off by default.
I think that we have that line for cases in which the "ground routing" (see what it is on #443) is used, but I never heard anybody saying that they are using it.

ggsnpdsn, 2023-04-18:
I'll let you know the test results when I'm free.

ilario, 2023-04-25:
? Is this the effect of removing that line??

Another thing that you can try is to use BATMAN_V instead of BATMAN_IV, see #1014

ggsnpdsn, 2023-04-27:
> ? Is this the effect of removing that line??
YES
> Another thing that you can try is to use BATMAN_V instead of BATMAN_IV, see #1014

BATMAN_V performance is worse, the problem still exists, and the forwarding rate is more unstable

ilario, 2024-11-07:
For speeding up the routing decisions, please try decreasing the time between Batman-adv hello packets (Originator Messages OGM), as documented here:
https://github.com/libremesh/lime-packages/blob/6642f373908d67c9cfc9d8f360843e9c528a8b94/packages/lime-docs/files/www/docs/lime-example.txt#L62

The current value is large as most networks are rather static.