libremesh/lime-packages #996: Fix category of babled-auto-gw-mode

aparcar, 2023-03-29:
It's in an entirely different category than (most) LibreMesh packages for no obvious reason

https://github.com/libremesh/lime-packages/blob/73e92cbb63e8d9ecb6660015e86b709b4b0ecacd/packages/babeld-auto-gw-mode/Makefile#L10

PalAman-git, 2023-04-07:
I want to contribute can you please guide me how to fix it

ilario, 2023-04-07:
Great!
Can you follow this guide? https://github.com/libremesh/lime-packages/blob/master/CONTRIBUTING.md
Then the only thing you should change is from:

```
  SECTION:=utils
  CATEGORY:=Utilities
```

to:
```
  SECTION:=lime
  CATEGORY:=LibreMesh
```

inside the `lime-packages/packages/babeld-auto-gw-mode/Makefile` file.
Thanks!

PalAman-git, 2023-04-07:
I have fixed the issue and requested the pull request. please review and merge it

ilario, 2023-04-07:
Fixed in #1006 
Thanks!

closed 2023-04-07 by ilario