libremesh/lime-packages #1002: Create module for using Opportunistic Wireless Encryption

ilario, 2023-04-05:
Opportunistic Wireless Encryption (OWE) is a kind of encryption of an AP that does not need of a password.
So people can have their traffic encrypted even if there is no authentication.
Clearly, this makes sense only if also the inter-nodes ieee802.11s mesh traffic is encrypted.

The documentation of its usage in OpenWrt is scarce:

* https://openwrt.org/docs/guide-user/network/wifi/basic#common_options1
* https://forum.openwrt.org/t/owe-and-transition-ssid/43502/2
* https://github.com/openwrt/openwrt/commit/574539ee2cdbb3dd54086423c6dfdd19bb1c06a6

Very likely, the roaming that we have now between open APs would not work.
So this should be activated only on the `apname` interfaces, for which the roaming does not happen.

ilario, 2023-05-27:
Doubt: if there is encryption, will the roaming work?
Would we need something like 802.11r (discussed in #253)?

Jadhav-Dhanashri, 2026-01-17:
Hello maintainers ,
I'm new to freifunk.net open source contribution & this would be my first issue. I'd like to work on this if it's available & learn from the process. Could you please assign it to me?
Thank you !

ilario, 2026-01-17:
> Hello maintainers , I'm new to freifunk.net open source contribution & this would be my first issue. I'd like to work on this if it's available & learn from the process. Could you please assign it to me? Thank you !

Please go ahead, it is available and nobody is working on this. Thanks for your help!