libremesh/lime-packages #1008: All routers answering to anygw IP when connected via LAN-LAN

ilario, 2023-04-12:
I first reported this issue here (first spotted by @pedro-nonfree) https://github.com/libremesh/lime-packages/issues/1007#issuecomment-1503684283 but @pony1k correctly identified that the issues are idependent.

What happens is that when one use the anygw IP (e.g. connecting to thisnode.info) gets answers from both routers, if they are connected in this way:

Laptop-ethernet port____lan1-router1-lan2____lan1-router2

This was observed using DSA-supported routers (PlasmaCloud PA1200 with both ports configured as LAN; Xiaomi MiRouter 4A gigabit edition) using OpenWrt 22.03 branch.

Seems that this should be avoided by these rules: 
https://github.com/libremesh/lime-packages/blob/8aa007e456d3cc98fd45997689624dbbeade3c7b/packages/lime-proto-anygw/files/etc/init.d/lime-anygw-ebtables#L9-L10

G10h4ck, 2023-04-14:
In case of non-DSA hardware switches, those rules does not even see the packets because they are switched directly by the switching chip without pushing them to the CPU and then to the kernel.

In case of DSA hardware switch it probably depends on the implementation, reading a bunch of kernel and driver code should help deciphering this.

G10h4ck, 2023-04-14:
anyway one should not use AnyGW IP for SSH or similar things, having the lime-app responding on that IP is already an usability compromise

LaneaLucy, 2023-04-19:
Shouldn't there be ground routing set, if you connect 2 lime systems over cable, which by default uses different vlans, which again solves the problem, that the CPU don't see the packets?

G10h4ck, 2023-04-21:
> Shouldn't there be ground routing set, if you connect 2 lime systems over cable, which by default uses different vlans, which again solves the problem, that the CPU don't see the packets?

That is a possibility but it doesn't happen automatically

ilario, 2024-11-07:
> anyway one should not use AnyGW IP for SSH or similar things, having the lime-app responding on that IP is already an usability compromise

@G10h4ck Also lime-app will have a very erratic behavior in this setup, no? Users are encouraged to access lime-app via names like thisnode.info, that gets resolved with the AnyGW IP. Then the browser will randomly receive answers from both router1 and router2, no?

Anyways this LAN-LAN setup causes other severe issues, like the one described in #1121 and #1118. In this setup a specific interface configuration is recommended, where the ethernet interface used for connecting to the other LibreMesh node is removed from the LAN bridge. I suppose this interface-specific configuration fixes this "multiple answers" problem also.

Please feel free to re-open if my assumption is wrong!!!

closed 2024-11-07 by ilario