libremesh/lime-packages #1019: Firmware that only supports Layer2 MESH

ggsnpdsn, 2023-04-24:
How to compile a firmware that only supports Layer2 MESH, and if there is a wired connection between two nodes, the two nodes are not allowed to support the establishment of a wireless link or prefer to use the wired link to forward data.

ilario, 2023-04-25:
I understand that these are two different questions:

For creating an image with only Layer 2 meshing, you can try to deselect lime-proto-babeld at compilation time. I never tried that and, as it is not officially supported, could have problems. Please let us know how it goes!

For having the preference of the connection using the cable I understand that is the same as in #993 ?

ilario, 2024-11-07:
> For creating an image with only Layer 2 meshing, you can try to deselect lime-proto-babeld at compilation time. I never tried that and, as it is not officially supported, could have problems.

@ggsnpdsn did you try? The discussion in this could go on on #468, so I close this, but please let us know if you tried and what you observed.

closed 2024-11-07 by ilario