libremesh/lime-packages #1034: Allow customization of checking frequency in wifi-unstuck-wa

ilario, 2023-07-24:
Currently, [wifi-unstuck-wa](https://github.com/libremesh/lime-packages/tree/ae6b33b6d874b473f8a4c6c087d88bdd4866cc7d/packages/wifi-unstuck-wa) is set to run (if installed, it is not in the recommended packages) every 10 minutes:

https://github.com/libremesh/lime-packages/blob/ae6b33b6d874b473f8a4c6c087d88bdd4866cc7d/packages/wifi-unstuck-wa/files/etc/uci-defaults/wifi-unstuck-cron#L9

And has an internal maximum limit to the execution time set to 5 minutes:

https://github.com/libremesh/lime-packages/blob/ae6b33b6d874b473f8a4c6c087d88bdd4866cc7d/packages/wifi-unstuck-wa/files/usr/lib/lua/wifi_unstuck_wa.lua#L12

It would be useful to be able to customize the 10 minutes time. For example for reducing it to 5 minutes and assuring that the link downtime will not be longer than 5 min.
Reported by @a-gave

pony1k, 2023-07-29:
I added
```
        interval = tonumber( config.get("wifi", "unstuck_interval", -1) )                                  
                                                                                                           
        if interval and interval > 0 then                                                                  
                --! use sed to replace interval in /etc/crontabs/root                                      
                io.popen("sed -i 's/\\*\\/\\d\\+ \\* \\* \\* \\* ((wifi-unstuck &> \\/dev\\/"..            
                "null)&)/*\\/"..interval.." * * * * ((wifi-unstuck \\&> \\/dev\\/null)\\&)/g"..            
                "' /etc/crontabs/root")                                                                    
        end 
```
on top inside the `function wireless.configure()` inside `/usr/lib/lua/lime/wireless.lua` and this now lets me customize the interval like so:
```
config lime 'wifi'
        option unstuck_interval '5'
        ...
```
Would that be an option?

ilario, 2023-07-31:
Yes!
I cannot think of a better way to do that.

@gothos does this work for you?

Can you make a pull request with that?
I would also include a line with the default value in the lime-system/files/etc/config/lime-defaults file and in the lime-docs/files/www/docs/lime-example.txt with some description.
Also, in the lime-example.txt we need to indicate the minimum time (that currently is the 5 minutes mentioned above, right?).
We could also consider decreasing the wu.TIMEOUT, but I let this decision to you.

Thanks!!!!

PR #1039 (2023-08-22): wifi-unstuck-wa: allow parametrizable values for interval and timeout
This close #1034
closed 2023-08-22 by a-gave