libremesh/lime-packages #1035: support config distribution accross the mesh

javierbrk, 2023-07-25:


ilario, 2023-07-25:
Eh?

https://ci6.googleusercontent.com/proxy/JQLv3rrAVtmdjtHJ_CuE8rvcTReP_c88YV5GSyxMLHlBBHTylUdXZLiq8cYah0fLlof1yZPxQYZAjcOwwEvf863OzR8AWXNYIwV4cS-gT07IgkVTYs6btQZgW4MngDuxHMEDUSHBiS02tdASnnMkfInMAfWlR48GvJdmVNMPFKC3P-nYaFrTgqCuqZuwb0hJSRWzA28EXvOnL12c7haLGJOevxsrnezZ_Nc=s0-d-e1-ft#https://github.com/notifications/beacon/AAKEO6WVJOKQ7IJ3BEMFJ4TXSAG4PA5CNFSM6AAAAAA2XPET22WGG33NNVSW45C7OR4XAZNFJFZXG5LFVJRW63LNMVXHIX3JMTHGZCIV4A.gif>Message
ID: ***@***.***>

javierbrk, 2023-07-26:
Sorry I didn't expand the idea in first place. It wold be a great feature to be able to spread community configurations from one node to all the others "automatically".  Most of those configurations are placed in each node in /etc/config/lime-community  https://libremesh.org/docs/es_config.html.

ilario, 2023-07-26:
Currently, each node with FirstBootWizard is exposing that file via the HTTP server (including the hash of the root password, if set with FirstBootWizard, see https://github.com/libremesh/lime-packages/issues/849), so that other nodes can fetch it.

You mean that the lime-community should be shared via shared-state?
What is the use case? Why?