libremesh/lime-packages #1049: Maintain compatibility with OpenWrt ImageBuilder and with the Attended Sysupgrade Server

a-gave, 2023-09-13:
Using the latest version of asu, the OpenWrt Attended Sysupgrade Server is possibile to build firmware images for LibreMesh.
It use the OpenWrt ImageBuilder to cook the firmware image. This basically merge a precompiled kernel and precompiled packages.
This process can be considered useful to build a libremesh image quickly, and without having a pre-existent setup of the OpenWrt Buildroot.
However this don't come without annoying bugs that maintain preferable having a consistent but slower openwrt buildroot.

Let's start to list these annoying issues:
- [ ] In any device that runs a libremesh firmware image produced with the imagebuilder the command `opkg list` return `Segmentation Fault`
  - [ ] This means you can install a plain openwrt and then use the asu clients - `auc` and `luci-app-attendedsysupgrade` - to build a libremesh image for that device. But you can't to it a second time from the device itself  (it could be possible selecting the `advanced mode` and inserting manually the list of packages, avoiding that that list is provided by `opkg list` , not yet tried) 

Example of request to build libremesh using asu:
```
{
  "packages": [
    "-dnsmasq",
    "-odhcpd_ipv6only",
    "profile-libremesh-suggested-packages"
  ],
  "repository_keys": [
    "RWSnGzyChavSiyQ+vLk3x7F0NqcLa4kKyXCdriThMhO78ldHgxGljM/8"
  ],
  "repositories": {
      "openwrt_core": "https://downloads.openwrt.org/releases/22.03.5/targets/ath79/generic/packages",
      "openwrt_base": "https://downloads.openwrt.org/releases/22.03.5/packages/mips_24kc/base",
      "openwrt_luci": "https://downloads.openwrt.org/releases/22.03.5/packages/mips_24kc/luci",
      "openwrt_packages": "https://downloads.openwrt.org/releases/22.03.5/packages/mips_24kc/packages",
      "openwrt_routing": "https://downloads.openwrt.org/releases/22.03.5/packages/mips_24kc/routing",
      "openwrt_telephony": "https://downloads.openwrt.org/releases/22.03.5/packages/mips_24kc/telephony",
      "libremesh": "https://feed.libremesh.org/master",
      "profile": "https://feed.libremesh.org/profiles"
  },
  "profile": "dlink_dir-842-c3",
  "target": "ath79/generic",
  "version": "22.03.5"
}
```

Example using curl
```
curl -X 'POST' \
  'https://sysupgrade.antennine.campiinrete.org/api/v1/build' \
  -H 'accept: application/json' \
  -H 'Content-Type: application/json' \
  -d '{
  "packages": [
    "-dnsmasq",
    "-odhcpd_ipv6only",
    "profile-libremesh-suggested-packages"
  ],
  "repository_keys": [
    "RWSnGzyChavSiyQ+vLk3x7F0NqcLa4kKyXCdriThMhO78ldHgxGljM/8"
  ],
  "repositories": {
      "openwrt_core": "https://downloads.openwrt.org/releases/22.03.5/targets/ath79/generic/packages",
      "openwrt_base": "https://downloads.openwrt.org/releases/22.03.5/packages/mips_24kc/base",
      "openwrt_luci": "https://downloads.openwrt.org/releases/22.03.5/packages/mips_24kc/luci",
      "openwrt_packages": "https://downloads.openwrt.org/releases/22.03.5/packages/mips_24kc/packages",
      "openwrt_routing": "https://downloads.openwrt.org/releases/22.03.5/packages/mips_24kc/routing",
      "openwrt_telephony": "https://downloads.openwrt.org/releases/22.03.5/packages/mips_24kc/telephony",
      "libremesh": "https://feed.libremesh.org/master",
      "profile": "https://feed.libremesh.org/profiles"
  },
  "profile": "dlink_dir-842-c3",
  "target": "ath79/generic",
  "version": "22.03.5"
}
'
```

ilario, 2023-09-13:
Oooh, this segmentation fault sounds like a big bug! On which version of OpenWrt did you observe this?
Is this already reported in OpenWrt issues tracker?

Regarding the general topic, check out also #1000

a-gave, 2023-09-14:
> Oooh, this segmentation fault sounds like a big bug! On which version of OpenWrt did you observe this?
Is this already reported in OpenWrt issues tracker?
It doesn't happen with plain openwrt, only on libremesh. 
To replicate follow the instructions at https://github.com/libremesh/lime-packages#using-the-imagebuilder-1
Encountered on: 
- openwrt-19.07.10-ar71xx-generic-ubnt-lbe-m5
- openwrt-22.03.5-ath79-generic-tplink_cpe510-v3
- (others with 22.03.5 then reflashed with an image produced by buildroot)

pony1k, 2023-11-23:
The segmentation fault happens when there is a package with `-` as version string. No packages in [http://feed.libremesh.org/master/](http://feed.libremesh.org/master/) have version `-` any more, so it seems like this has been fixed now.
I have router with a recent image made with imagebuilder and libremesh and `opkg list` works fine. For the older ones I had to add `Version: unknown` to the affected packages in `/etc/opkg/status` to make it work.

ilario, 2024-11-07:
Closing, as it seems already fixed. Please re-open if it still happens.

closed 2024-11-07 by ilario