libremesh/lime-packages #1058: Need Support for Two Different Networks in the Mesh

DaviReisVieira, 2023-10-09:
Description: I am currently working with Libre-Mesh and I have a requirement to set up two different networks within the mesh. This is important for my project because I need to isolate certain devices and services from each other while still using the same physical mesh infrastructure.

Expected Behavior: I would like to see support or guidance on how to configure and manage two distinct networks within the Libre-Mesh framework. It would be great if there could be clear documentation or examples to help achieve this.

Additional Information:

- I am using the latest version of Libre-Mesh.
- I have already reviewed the existing documentation and couldn't find information specific to creating multiple networks.
- Any advice, tips, or pointers from experienced users or developers would be highly appreciated.

Please let me know if you need any further information or clarification. Thank you for your support!

ilario, 2023-10-11:
Thanks for the question @DaviReisVieira, it is certainly a very useful network setup, but I never tried to achieve it.

Maybe someone else? @spiccinini @G10h4ck @nicoechaniz @digitigrafo @a-gave @altergui @dangowrt @javierbrk @pony1k @aparcar 

In the meanwhile, you can set up a VPN between the clients of one half of the network, the most sensitive one. Without doing any special configuration on the mesh nodes, setting up the VPN only on the clients to be kept isolated.

javierbrk, 2023-10-13:
Hi, I am not an expert, but I've been doing a review of the BATMAN protocol
and found this article.

 https://cgomesu.com/blog/Mesh-networking-openwrt-batman/#mesh-vlans

It has a specific section about vlans. The article is very didactic and
clear. Hopefully you will find it useful.

Best regards
Javier

ilario, 2024-11-07:
@DaviReisVieira do you have any update on this? Did you find a solution?
Did the Batman-adv VLAN solution proposed by @javierbrk helped?