libremesh/lime-packages #1060: lime-proto-lan confuses bridge device section

pony1k, 2023-10-17:
After installing OpenWrt 23.05.0 with current master branch (https://github.com/libremesh/lime-packages/commit/4c51c7e062c3dff6c5218ee797d5a79c2ad6bc3d) onto an [avm fritz4020](https://openwrt.org/toh/avm/fritz.box.4020#avm_fritz_box_4020) I found the following in `/etc/config/network`:
```
config device                               
        option name 'eth1'                                
        option macaddr '44:4E:6D:11:87:D0'                
        list ports 'bat0'                   
        list ports 'eth0.1'                               
        list ports 'eth1'      
```
The ports should be added to the the bridge device further down. It probably has something to do with this line:
https://github.com/libremesh/lime-packages/blob/4c51c7e062c3dff6c5218ee797d5a79c2ad6bc3d/packages/lime-system/files/usr/lib/lua/lime/proto/lan.lua#L51-L53

pony1k, 2023-10-17:
Moving the device section with `option type 'bridge'` before the `eth1` device section fixed it for me.

ilario, 2023-10-18:
Argh, looks like a big assumption we made is failing.
We assumed that the bridge is always the first device in the network file.
We assumed that because that section has no name, so it is difficult to address it properly. An alternative would be to check which section has the type bridge, I suppose.

Maybe someone from OpenWrt can suggest how to do that the right way?
@hauke @dangowrt @aparcar @BKPepe

PR #1061 (2023-10-22): draft: fix bridge device section confusion
Bridge ports are added to the first device section, which may be the wrong one. This changes the behaviour follwing @ilario s [suggestion](https://github.com/libremesh/lime-packages/issues/1060#issuecomment-1767832137). Now, the first device section that has type `bridge` and the name `br-lan` will be used.
After reading the `/bin/config_generate` script in OpenWrt base-files I am confident that there will always be such a device section on first boot.

fixes #1060
closed 2023-10-22 by spiccinini