libremesh/lime-packages #1063: wifi-band confusion on TP-LINK TD-W8970

pony1k, 2023-11-14:
`TP-LINK TD-W8970` is a 2GHz only device, but lime-config configures the radio as 5GHz, which does not work. The cause of the problem is that in `wireless.lua` the function
```
function wireless.is5Ghz(radio)
	local devModes = iwinfo.nl80211.hwmodelist(radio)
	return devModes.a or devModes.ac
end
```
returns `true` for `radio0`. I investigated on why this is, like so:
```
root@device:~# lua
Lua 5.1.5  Copyright (C) 1994-2012 Lua.org, PUC-Rio (double int32)
> iwinfo = require("iwinfo")
> devModes = iwinfo.nl80211.hwmodelist('radio0')  
> for k,v in pairs(devModes) do
>> print(k,v)
>> end
a       true
ac      false
b       true
ad      false
ax      false
g       true
n       true
```
Interestingly it returns `true` for `a`. Also interesting is the output of `iw phy phy0 info`:
```
Wiphy phy0
	wiphy index: 0
	max # scan SSIDs: 4
	max scan IEs length: 2257 bytes
	max # sched scan SSIDs: 0
	max # match sets: 0
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 23 (up to 10350m)
	Device supports AP-side u-APSD.
	Device supports T-DLS.
	Available Antennas: TX 0x7 RX 0x7
	Configured Antennas: TX 0x7 RX 0x7
	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * AP/VLAN
		 * monitor
		 * mesh point
		 * P2P-client
		 * P2P-GO
		 * outside context of a BSS
	Band 1:
		Capabilities: 0x11ef
			RX LDPC
			HT20/HT40
			SM Power Save disabled
			RX HT20 SGI
			RX HT40 SGI
			TX STBC
			RX STBC 1-stream
			Max AMSDU length: 3839 bytes
			DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: 8 usec (0x06)
		HT TX/RX MCS rate indexes supported: 0-23
		Frequencies:
			* 2412 MHz [1] (20.0 dBm)
			* 2417 MHz [2] (20.0 dBm)
			* 2422 MHz [3] (20.0 dBm)
			* 2427 MHz [4] (20.0 dBm)
			* 2432 MHz [5] (20.0 dBm)
			* 2437 MHz [6] (20.0 dBm)
			* 2442 MHz [7] (20.0 dBm)
			* 2447 MHz [8] (20.0 dBm)
			* 2452 MHz [9] (20.0 dBm)
			* 2457 MHz [10] (20.0 dBm)
			* 2462 MHz [11] (20.0 dBm)
			* 2467 MHz [12] (20.0 dBm)
			* 2472 MHz [13] (20.0 dBm)
			* 2484 MHz [14] (disabled)
	Band 2:
		Capabilities: 0x11ef
			RX LDPC
			HT20/HT40
			SM Power Save disabled
			RX HT20 SGI
			RX HT40 SGI
			TX STBC
			RX STBC 1-stream
			Max AMSDU length: 3839 bytes
			DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: 8 usec (0x06)
		HT TX/RX MCS rate indexes supported: 0-23
		Frequencies:
			* 5180 MHz [36] (disabled)
			* 5200 MHz [40] (disabled)
			* 5220 MHz [44] (disabled)
			* 5240 MHz [48] (disabled)
			* 5260 MHz [52] (disabled)
			* 5280 MHz [56] (disabled)
			* 5300 MHz [60] (disabled)
			* 5320 MHz [64] (disabled)
			* 5500 MHz [100] (disabled)
			* 5520 MHz [104] (disabled)
			* 5540 MHz [108] (disabled)
			* 5560 MHz [112] (disabled)
			* 5580 MHz [116] (disabled)
			* 5600 MHz [120] (disabled)
			* 5620 MHz [124] (disabled)
			* 5640 MHz [128] (disabled)
			* 5660 MHz [132] (disabled)
			* 5680 MHz [136] (disabled)
			* 5700 MHz [140] (disabled)
			* 5745 MHz [149] (disabled)
			* 5765 MHz [153] (disabled)
			* 5785 MHz [157] (disabled)
			* 5805 MHz [161] (disabled)
			* 5825 MHz [165] (disabled)
	valid interface combinations:
		 * #{ managed } <= 2048, #{ AP, mesh point } <= 8, #{ P2P-client, P2P-GO } <= 1, #{ IBSS } <= 1,
		   total <= 2048, #channels <= 1, STA/AP BI must match, radar detect widths: { 20 MHz (no HT), 20 MHz, 40 MHz }

	HT Capability overrides:
		 * MCS: ff ff ff ff ff ff ff ff ff ff
		 * maximum A-MSDU length
		 * supported channel width
		 * short GI for 40 MHz
		 * max A-MPDU length exponent
		 * min MPDU start spacing
	max # scan plans: 1
	max scan plan interval: -1
	max scan plan iterations: 0
	Supported extended features:
		* [ RRM ]: RRM
		* [ FILS_STA ]: STA FILS (Fast Initial Link Setup)
		* [ CQM_RSSI_LIST ]: multiple CQM_RSSI_THOLD records
		* [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
		* [ TXQS ]: FQ-CoDel-enabled intermediate TXQs
		* [ AIRTIME_FAIRNESS ]: airtime fairness scheduling
		* [ SCAN_RANDOM_SN ]: use random sequence numbers in scans
		* [ SCAN_MIN_PREQ_CONTENT ]: use probe request with only rate IEs in scans
		* [ CAN_REPLACE_PTK0 ]: can safely replace PTK 0 when rekeying
		* [ CONTROL_PORT_NO_PREAUTH ]: disable pre-auth over nl80211 control port support
		* [ DEL_IBSS_STA ]: deletion of IBSS station support
		* [ MULTICAST_REGISTRATIONS ]: mgmt frame registration for multicast
		* [ SCAN_FREQ_KHZ ]: scan on kHz frequency support
		* [ CONTROL_PORT_OVER_NL80211_TX_STATUS ]: tx status for nl80211 control port support
```
It looks like it supports 5Ghz, but is somehow 'disabled'. OpenWrt version is `22.03.5`. A workaround is to configure a 2.4GHz channel for the 5GHz band.

a-gave, 2023-11-14:
It seems to me it could be a problem related to how this specific device is mapped inside openwrt, more than a libremesh issue:

This device share the same vr9_tplink_tdw89x0.dtsi with vr9_tplink_tdw8980.dts

It seems it was firstly an error of the manufacturer, fixed in:
https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=ec21c9821c5c8cb2ce7af68dbd497fe9c9c60667

Remanaged then in:
https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=4c8dd973ef8e76f9fc622f0baab981769c89b601
"Instead of disabling the
drivers logic which would add the affected band and
channels. It now disables all channels which are not
within the specified frequency range."

Double check also these if you can:
This is the history of the dtsi (after the renaming to soc_vendor_device):
https://git.openwrt.org/?p=openwrt/openwrt.git;a=history;f=target/linux/lantiq/files/arch/mips/boot/dts/lantiq/vr9_tplink_tdw89x0.dtsi;h=5b7c10bb9dd49b5a3959a8818015bee06ea93da4;hb=HEAD

This the history (before renaming to soc_vendor_device )
https://git.openwrt.org/?p=openwrt/openwrt.git;a=history;f=target/linux/lantiq/files/arch/mips/boot/dts/TDW89X0.dtsi;h=eabbc0257f92d8f37665cb7277295dff9a9fd98f;hb=de6dd7a8db1c073734113156668bca6792dc123d

So for me from it could be considered a normal behaviour in openwrt, 
From the libremesh side, I would try to stop trying to configure both a 2.4GHz radio and a 5ghz radio
And I would try deselecting all 5ghz related configs from `/etc/config/lime-default` and see if the 2.4GHz is up

	# config lime-wifi-band '5ghz'
	#	list channel '48'
	#	list channel '157'
	#	option htmode 'HT40'
	#	option distance '10000'
	#	option adhoc_mcast_rate '6000'
	#	option ieee80211s_mcast_rate '6000'

Let us know if it is of any help

pony1k, 2023-11-15:
In `/lib/wifi/mac80211.sh` (provided  by `kmod-cfg80211`), there is a function `__get_band_defaults()` which is used by OpenWrt to correctly detect the wifi band of a phy. It just looks at the lines `Band <x>` in the output of `iw phy <phyname> info`, where `<x>` is `1` for 2g and `2` for 5g. It ignores bands without enabled channels. Maybe LibreMesh could do the same?

ilario, 2023-11-16:
Wow sounds good!
We could "source" (execute) that file and use its function.
No idea if this is easy to do in Lua...

G10h4ck, 2024-01-23:
> In `/lib/wifi/mac80211.sh` (provided by `kmod-cfg80211`), there is a function `__get_band_defaults()` which is used by OpenWrt to correctly detect the wifi band of a phy. It just looks at the lines [...]

It seems `wireless.is5Ghz` should be re-implemented on top of that function, @pony1k would you mind to make a PR for that?
 ;)

pony1k, 2024-01-24:
> It seems `wireless.is5Ghz` should be re-implemented on top of that function

This would work for all radios except broadcom radios. As far as I understand, broadcom radios are not compatible with cfg80211 subsystem. For those radios [/lib/wifi/broadcom.sh](https://github.com/openwrt/openwrt/blob/3cf1fe5508ee8a2a2c6e33a829bfb6c81381ccd0/package/kernel/broadcom-wl/files/lib/wifi/broadcom.sh) is used instead of [/lib/wifi/mac80211.sh](https://github.com/openwrt/openwrt/blob/3cf1fe5508ee8a2a2c6e33a829bfb6c81381ccd0/package/kernel/mac80211/files/lib/wifi/mac80211.sh).
The best solution I can think of right now is to just look at the `option band ...` in the `config wifi-iface ...`, which is populated with the the value returned by `__get_band_defaults()`.

G10h4ck, 2024-01-25:
@pony1k do broadcom radios

PR #1071 (2024-01-25): lime.wireless: Fix wireless.is5Ghz
Fix #1063, where wireless.is5Ghz returns true for the 2GHz radio on TP-LINK TD-W8970.
commit 0c5b5d5f (2024-01-25): lime.wireless: Fix wireless.is5Ghz

Fix #1063, where wireless.is5Ghz returns true for the 2GHz radio on
TP-LINK TD-W8970.
closed 2024-01-25 by G10h4ck