libremesh/lime-packages #1075: lime-app still uses $RANDOM in crontab

pony1k, 2024-02-02:
These two scripts add lines to crontab that use ash `$RANDOM`, which does not work in OpenWrt 19.07 to OpenWrt 22.03.

- [shared-state-reference_state_nodes_info_cron](https://github.com/libremesh/lime-packages/blob/5053cf0cb46b7f865d9c3917c3b08232c88b66d1/packages/lime-app/files/etc/uci-defaults/shared-state-reference_state_nodes_info_cron)
- [shared-state-reference_state_wifi_links_info_cron](https://github.com/libremesh/lime-packages/blob/5053cf0cb46b7f865d9c3917c3b08232c88b66d1/packages/lime-app/files/etc/uci-defaults/shared-state-reference_state_wifi_links_info_cron)

Are these cronjobs still needed? Should they be repaired?

ilario, 2024-04-04:
I start thinking that we have to abandon OpenWrt 22 and previous ones, as shared-state-async is not possible to compile on them.
And $RANDOM is available on OpenWrt 23 (https://github.com/libremesh/lime-packages/issues/800#issuecomment-1522950082) so we should eliminate random_numgen at some point and use only $RANDOM.
Do you agree?

pony1k, 2024-04-05:
I agree. I can see no reason to use OpenWrt 22 or earlier on new installations.

PR #1117 (2024-08-01): Stop using random-numgen, not needed in recent OpenWrt
The random-numgen package was created in #991 as in OpenWrt 21 BusyBox did not have $RANDOM enabled by default (so it was not present in the official ImageBuilder), see #800.

But since OpenWrt 22 it is present by default, see https://github.com/openwrt/packages/pull/20728 so there is no need to use it.

Maybe we can keep the random-numgen package for a few more years and then delete it.

This commit should be picked also in the 2024.1 branch.

Fix #1075
closed 2024-08-01 by G10h4ck