libremesh/lime-packages #1093: Bat-hosts get_bathost error

selankon, 2024-03-11:
Trying libremesh with OWRT23, me and other people noticed that bat host doesn't load correctly a bathost data:

```
root@LiMe-eb7f64:~# ubus  call bat-hosts get_bathost '{"mac": "D2:5F:25:EB:63:DA", "outgoing_i
face": "wlan1-mesh" }'
Command failed: No response
root@LiMe-eb7f64:~# echo '{"mac": "D2:5F:25:EB:63:DA", "outgoing_iface": "wlan1-mesh" }' | /us
r/libexec/rpcd/bat-hosts call get_bathost | jq
Failed to allocate uclient context
lua: /usr/libexec/rpcd/bat-hosts:31: attempt to index local 'bathost' (a nil value)
stack traceback:
	/usr/libexec/rpcd/bat-hosts:31: in function 'get_bathost'
	/usr/libexec/rpcd/bat-hosts:52: in main chunk
	[C]: ?
```

It seems that is not retrieving the info or nill values are not handled properly.  

This causes a "Fetching name" loader continuously shown on the lime app landing page.

ilario, 2024-03-14:
Who can investigate this?
@javierajorge can you have a look? Thanks!

ilario, 2024-11-07:
According to this comment
https://github.com/libremesh/lime-packages/pull/1094#issuecomment-2000385170 this has been fixed. Please @javierbrk @selankon open it again if I misunderstood.

closed 2024-11-07 by ilario