libremesh/lime-packages #1119: Enable WAN and share internet to the virtual machine.

Lucho-rar, 2024-07-17:
**Description:**

Hello! My name is Luciano, and I am a beginner in the concepts of LibreMesh and OpenWRT.

I compiled the OpenWRT source code with the LiMe packages following the guide provided in [LibreMesh Development Guide](https://libremesh.org/development.html). However, I encountered issues when trying to share my local host connection with the emulated instance in QEMU using the `--enable-wan` flag.

After some investigation and with @javierbrk help, I managed to resolve this issue by disabling the `systemd-resolved` service before starting the emulation and then re-enabling it afterward. Specifically, I ran the following commands in the `Linux Mint` context:

```bash
$ sudo systemctl disable systemd-resolved --now
$ sudo ./tools/qemu_dev_start  path/to/openwrt-x86-64-generic-rootfs.tar.gz path/to/openwrt-x86-64-ramfs.bzImage
$ sudo systemctl enable systemd-resolved --now
```

**Contribution proposal:**

My proposal is to update the documentation in [TESTING.md](https://github.com/libremesh/lime-packages/blob/master/TESTING.md) to include this error among the possible errors, if it helps.

Thank you for your time and for maintaining this amazing project.

Sincerely,
Luciano

ilario, 2024-07-18:
Thanks! Unexpected!
Rather than `disable --now` and `enable --now`, would `stop` and `start` be enough?

Also, I think that this content should go to VIRTUALIZING.md (a draft of this file is in #938, we should merge it) and also all the stuff about qemu from TESTING.md should be moved to VIRTUALIZING.md, in my opinion.

Lucho-rar, 2024-07-18:
Thanks for your feedback, ilario!

I tested using `stop` and `start` and it works as expected. I will update my steps accordingly.

Regarding the suggestion to move the content to VIRTUALIZING.md and shift the QEMU related information from TESTING.md, I'm happy to contribute with that if you think it’s appropriate.

Thanks again for the guidance!

ilario, 2024-11-07:
Yes please, expand the documentation :D