libremesh/lime-packages #1120: Own IPv6 broadcast domain

ilario, 2024-07-26:
@pony1k proposed to have a LibreMesh-specific IPv6 broadcast domain.

Background:
Now one can do `ping6 ff02::1%name_of_interface` and get answers from all directly reachable devices.

Proposal:
One could ping another address for getting answers from all and only the LibreMesh devices, something like `ping6 ff12::bee%name_of_interface`

This would be convenient for general network maintenance, and specifically it would help shared-state to detect other LibreMesh nodes replacing the generic IPv6 broadcast address here:
https://github.com/libremesh/lime-packages/blob/4c51c7e062c3dff6c5218ee797d5a79c2ad6bc3d/packages/shared-state/files/usr/bin/shared-state-get_candidates_neigh#L36-L65

The original proposal here, for easing the identification of the devices reachable from a specific interface:
https://lists.autistici.org/message/20240621.153757.1eab7dbb.en.html

A proposal of implementation here:
https://lists.autistici.org/message/20240726.154215.73440c31.en.html

pony1k, 2024-07-26:
I also wanted to share the idea that shared-state could use multicast-addresses for communication. Currently, it uses multicast ping to discover neighbors, then communicates with them via TCP. But I'm not qualified to decide whether that makes sense.

ilario, 2024-11-07:
@javierbrk @G10h4ck opinions?