libremesh/lime-packages #1121: Problems with ethernet connections between routers in default configuration

pony1k, 2024-07-31:
Continuing #1118 here because I'm not sure if trying to detect switches between routers is the right way to go forward.

Let's discuss the issue described by @ilario in this mail:
https://lists.autistici.org/message/20240714.140352.58fe57b2.en.html
 
In default configuration, ethernet interfaces are added to the br-lan bridge, while also being configured as batadv hard interface. In some setups, this leads to error messages appearing in the kernel log in a high rate and network instability.

It was suspected that the error appears iff there is a switch between the two routers. I tried to reproduce the issue with a dumb switch, but without success (everything working fine, no errors in kernel log), as described in this mail:
https://lists.autistici.org/message/20240726.150840.dcc0e028.en.html

I then tried to reprduce it by replacing the switch with an OpenWrt-router (without DSA), basicly acting as a managable switch, with no sucess either.

Then when I connected the two LibreMesh routers directly, suprisingly I could observe the issue. I could observe the error messages in the kernel logs and batadv didn't mesh over ethernet. On `mr70x-v1`, `batctl n` did not list the `fritz4040` as neighbour on the lan interfaces, also `batctl bbt` showed no routers in the backbone table on both routers. `batctl tcpdump lan1_29` could see batman OGMs appearing on the `lan1_29`, not sure why the interface was not showing up in the neighbour table. After a while, the wifi connection between my laptop and the routers became quite unusable. When I ran tcpdump on the mesh interfaces I found that there was a lot of broadcast and some frames were duplicated many times (I saw ICMPv6 messages with  same id and seq-no many times over long time periods. Plus, on my laptop, I saw the same echo request being received over and over at a high rate. So there was a loop and that clogged the wifi interface.

It is not the kind of loop I described in #1032 .

Later I booted the routers again, to further investigate the issue. Annoyingly, everything is working fine now. No kernel logs, meshing over ethernet works, no frames looping around. I'm not able to replicate the issue again. I also tried with resetting the configuration to firstboot state, but to no avail. So, unfortunatly, it is currently not possible for me to find out when excactly this happens and why.

I find it strange that batman is also configured on `eth0` on dsa enabled devices. I don't think we are supposed to use that directly. Next time someone observes this issue, maybe they could add
```
config net
	option linux_name 'eth0'
	list protocols 'manual'
```
to the `lime-node` file and see if it helps.

ilario, 2024-08-05:
Maybe some race condition is happening here?
Some time ago I remember there was something like the wrong interface being added to the bridge as the first one setting its MAC address to some harmful value, but I also think this was solved, maybe adding the dummy0 interface, cannot remember (and I found a message of  mine mentioning that we could remove dummy0 altogether as it should not be needed anymore https://github.com/libremesh/lime-packages/issues/189#issuecomment-445493854).

ilario, 2024-08-06:
Amazing, you have a great memory :D

ilario, 2024-11-07:
I believe that adding some documentation about how to configure the routers in this scenario can mitigate this issue, do you agree?

So, for the upcoming release we should either tackle this properly or (simpler) add comments in the website (on lime-docs' /docs/lime-example.txt @pony1k already added documentation in https://github.com/libremesh/lime-packages/pull/1085 ) indicating how to manage this.

ilario, 2024-11-07:
Also, this interface-specific configuration should be exposed via lime-app, as it is veeery common for users to connect two LibreMesh devices via ethernet (I believe).
Opinions? @selankon @javierbrk @G10h4ck

ilario, 2024-11-07:
The issue #1008 is like a child of this one. I hope that when this one will be fixed, also #1008 will really be fixed.

ilario, 2024-12-07:
The default configuration aims to support both LiMeRouter-Client and LiMeRouter-LiMeRouter ethernet connections.
While this maybe worked for both scenarios in the past, currently it works only for the first case, so why don't we change the default configuration?

Should we leave the Ethernet interfaces out of Batman-adv's bat0, or also out of Babeld?

What is a clean way to do so?
We should allow the users to activate it manually in interface-specific configuration.

There is already something like this in place for AP interfaces:

https://github.com/libremesh/lime-packages/blob/396c37ac8b853156e8c72dd51cb387736f4dc05c/packages/lime-proto-batadv/files/usr/lib/lua/lime/proto/batadv.lua#L53-L54

If we add there an exception also for cabled interfaces, it will disable Batman for LAN interfaces but it would still be present in WAN interfaces (as they get configurated with interface-specific configuration by [lime-hwd-openwrt-wan](https://github.com/libremesh/lime-packages/blob/master/packages/lime-hwd-openwrt-wan/) that should be ok, no?

ilario, 2024-12-07:
In case we should remove also Babeld, we can add the same check here:

https://github.com/libremesh/lime-packages/blob/c3ef5a80929dee895be3661a60e8b700324e4e5b/packages/lime-proto-babeld/files/usr/lib/lua/lime/proto/babeld.lua#L95-L97

pony1k, 2024-12-08:
> Should we leave the Ethernet interfaces out of Batman-adv's bat0, or also out of Babeld?

TL;DR  For what it's worth, I think this is a step in the right direction.

Unfortunately, I did not have time to do further testing. However, I think the reason that we are seeing this issue is that this kind of configuration is not supported by DSA. Let me explain. In the default configuration, we get something like this:
```
config device
    option name 'br-lan'
    option type 'bridge'
    list ports 'bat0'
    list ports 'lan1'
    list ports 'lan2'
    list ports 'lan3'
    
[...]

config device 'lm_net_lan1_batadv_dev'
    option type '8021ad'
    option name 'lan1_207'
    option ifname 'lan1'
    option vid '207'
    option macaddr '02:4a:18:35:7e:35'
    option mtu '1532'
```

The first section adds lan1 to br-lan, putting the DSA-port in bridge mode. The second section configures a soft vlan on top of lan1. That would require the port to be in stand-alone mode. But a DSA port can not be both in stand-alone and in bridge mode. What we want to happen is that ingress packets on lan1 that have a vlan tag of 207 should go to lan1_207, while all others should be handled by the bridge. This worked fine on swconfig devices, where the bridging is done in software. But with DSA-devices, the bridging is offloaded to the internal switch.

Not automatically configuring batadv and babeld on DSA-ports would avoid the situation where a DSA-port is both member of a bridge and configured as stand-alone port.

Maybe in future, node-to-node connections could be made possble again without manual configuration by cascading two bridges: One with all the DSA-ports and another bridge that has the first bridge as member, that we call br-lan and also has bat0 and the wlan-ap interfaces. Then the routing protocols could be configured on top of the first bridge. If we switch the tagging protocol on ethernet to 802.1q, then we could use bridge-vlans to seperate traffic between each routing protocol and user traffic. 

Here is some very detailed information on DSA:
https://www.kernel.org/doc/html/latest/networking/dsa/dsa.html
https://www.kernel.org/doc/html/latest/networking/dsa/configuration.html

ilario, 2024-12-09:
@pony1k amazing research!!

The only thing I can say (not very useful though) is that we can already use VLAN 802.1q just specifying it in the /etc/config/lime-node with something like:

`list protocols "batadv:%N1:8021q`

As implemented here:
https://github.com/libremesh/lime-packages/blob/396c37ac8b853156e8c72dd51cb387736f4dc05c/packages/lime-proto-batadv/files/usr/lib/lua/lime/proto/batadv.lua#L63-L64

And documented here:
https://github.com/libremesh/lime-packages/blob/396c37ac8b853156e8c72dd51cb387736f4dc05c/packages/lime-docs/files/www/docs/lime-example.txt#L42

If I remember correctly, @G10h4ck said we are using 802.1ad specifically for avoiding having the hardware switch messing with our packets. 

@G10h4ck pls have a look

pony1k, 2026-10-24:
I think I've found a promising hack to make the default configuration work again both for users and other nodes connected via ethernet.

To summarize the problem we are trying to solve:
- We can not add dsa ports to br-lan and then also configure .1ad soft vlan interfaces on top of single dsa ports, because the switch is not aware of the .1ad soft vlan and therefore does not know whether it should forward an incoming packet to the bridge or to the single port interface. This is how it was configured originally and the reason why this issue was opened.
- We can not add dsa ports to br-lan and then configure batadv on top of a .1ad vlan on top of br-lan, because batadv doesn't let us do that.
- We can not put all dsa ports in a separate bridge, define .1ad vlans on top of that and also add it to br-lan, because `Error: Can not enslave a bridge to a bridge.`
- We can not put all dsa ports in a separate bridge, define a single untagged bridge-vlan on top of it and define .1ad vlans on top of the bridge-vlan interface and put the bridge-vlan interface into br-lan. This seems to be working on some switches (like where I tested here: https://github.com/libremesh/lime-packages/pull/1203#issuecomment-3084072758). But it does not work on others (like @gothos tested here: https://github.com/libremesh/lime-packages/pull/1203#issuecomment-3137060000). I reproduced the same result gothos had on yet another router. The switch seems to drop incoming .1ad packets instead of treating them as untagged and forward them as-is. Maybe this could also happen with some swconfig devices, though I haven't seen this yet.

But what seems to work is the following: Define a macvlan in passthru mode on every dsa port. Passthru mode basically forwards all packets no matter the dest mac address to the upper interface, so we can add it to a bridge. There is an undocumented option `mode` in uci `config 'device'` sections with `option type 'macvlan'` that we can set to `passthru`. Since the macvlan interface is not a dsa port anymore, there is no offloading to the internal switch. So we can use the macvlan interfaces in place of the dsa ports and configure everything as we used to before https://github.com/libremesh/lime-packages/commit/13088ba6dfe69466e40a3567ee08a595bfe9cdbf was merged. I tested it on two cable connected devices Linksys E8450 and TP-Link RE650 v1. Both batadv and communication with the other devices br-lan bridge work. `batctl bbt` each shows the other device, so [BLA](https://www.open-mesh.org/projects/batman-adv/wiki/Bridge-loop-avoidance) seems to be working as expected. 

I can prepare a pull request with this hack in the next days, so others can test this more easily.

If this turns out to work reliably, I propose we use that for the short term and for backward compatibility. But for the long term I think we should switch to normal .1q vlans. This however requires more work, since we would need to start configuring switch-vlans and bridge-vlans.

a-gave, 2026-10-24:
Thanks a LOT, I found very useful also the article that you posted in the last meeting about these tools https://cnly.github.io/2018/11/09/conflicts-and-limitations-of-bridge-and-macvlan-devices.html

I tested this in this way in these routers: 
gemtek_wvrtm-127acn (dsa) ---cable--- openwrt_one (swconfig) ---cable--- cudy-wr3000s-v1 (dsa)

1. Create the macvlans on top of dsa lan ports (repeat for more ports lan1 lan2 lan3 lan4)
AND add them to the br-lan bridge, so they'll have an interface and get created during network restart
AND disable the ipv6 link local on the 'raw' dsa lan ports
```
config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'bat0'
	list ports 'lan1x'
	list ports 'lan2x'

config device 'lan1x'
	option type 'macvlan'
	option mode 'passthru'
	option name 'lan1x'
	option ifname 'lan1'

config device 'lan2x'
	option type 'macvlan'
	option mode 'passthru'
	option name 'lan2x'
	option ifname 'lan2'

config device 'lan1'
	option ifname 'lan1'
	option ipv6 '0'

config device 'lan2'
	option ifname 'lan2'
	option ipv6 '0'
```
3. reboot or reload to apply `/etc/init.d/network reload; /etc/init.d/network/restart`
4. Edit lime-node adding a section for each port and set manual to the lan ports so they aren't added to br-lan
```
config net 
	option linux_name 'lan1'
	list protocols 'manual'

config net 
	option linux_name 'lan2'
	list protocols 'manual'

config net
	option linux_name 'lan1x'
	list protocols 'babeld:17'
	list protocols 'batadv:%N1'
	list protocols 'lan'

config net
	option linux_name 'lan2x'
	list protocols 'babeld:17'
	list protocols 'batadv:%N1'
	list protocols 'lan'
```
5. run `lime-config; lime-apply`

Results:
- Connection between routers works reliably, no error log in dmesg and logread:
- Ok with: gemtek_wvrtm-127acn (dsa) ---cable--- openwrt_one (swconfig)
- OK with: gemtek_wvrtm-127acn (dsa) ---cable--- openwrt_one (swconfig) ---cable--- cudy-wr3000s-v1 (dsa)
- OK with gemtek_wvrtm-127acn (dsa) ---cable---  cudy-wr3000s-v1 (dsa) ---cable--- openwrt_one (swconfig)
- With this setup both dsa lan ports and their respective macvlan (i.e. lan1 and lan1x) have the same macaddress

Will test further and I would be very happy to test the pr!

a-gave, 2026-10-24:
Small further notes:
1. an host connected via cable reaches the anygw address of the correct node and now anygw it works properly always
2. The above configuration works both if .1ad interfaces are created on the macvlan lan1x and on the network device lan1
3. It seems not present anymore the issue (mentioned in #1210) of the ghost neighbour, where, given the connection below, the first node sees the last as directly connected (and vice versa), both in babeld and batadv .1ad) while they have only the second as direct neighbour.
gemtek_wvrtm-127acn (dsa) ---cable--- cudy-wr3000s-v1 (dsa) ---cable--- openwrt_one (swconfig)