libremesh/lime-packages #1122: Watchping no default config for wwan

LaneaLucy, 2024-08-15:
If i want to connect a libremesh node to the Internet via wifi, for example because my hardware has no wan port, i use luci, and when luci creates a wifi client, it name default is wwan. But watching only has default config for wan and wan6.
It would really help, when watching default config could also include config for wwan interface.

Thx

javierbrk, 2024-08-22:
When using lime-app there is an option to "connect to a mobile hotspot" just enable this option and configure your wifi access point as specified in the UI. Please let me know if this helps.

ilario, 2024-11-07:
The configuration made with LuCI is likely not going to be compatible with the LibreMesh stuff.

The supported configuration method are:

1) editing the /etc/config/lime-node adding a client like this:
https://github.com/libremesh/lime-packages/blob/v2024.1-rc1/packages/lime-docs/files/www/docs/lime-example.txt#L158-L177

2) using lime-app, as mentioned by @javierbrk 

Can you try with these two methods? I hope they work, but it would be great if you could confirm that :D