libremesh/lime-packages #1124: SharedState change node location problems

selankon, 2024-09-04:
If I change the node location after shared state starts, it doesn't update properly on shared state.

[image]
 
Another thing is that may have no sense to store the location on reference state. We could just use the current location all the time. There is no use case where a user deliberately changes the location and expect to see the reference state location and the new location on the map, imo

javierbrk, 2024-09-18:
the user should set a new reference state for the link, that could fix the problem.

ilario, 2024-11-07:
Does this affect the 2024.1-rc1 release candidate?