libremesh/lime-packages #1130: Firmware file not found error

selankon, 2024-09-17:
If you don't confirm, the firmware is supposed to be rollback.

However after reboot the error shown is `firmware_file_not_found`. Is this expected? 

[image]

Either, abort won't work because shared state is to slow

selankon, 2024-09-17:
Abort works when shared state has enough info

javierbrk, 2024-09-18:
yes.
when the scheduled is performed, the status is stored in the old partition and in the new partition. 
When you reboot into the old partition after **not confirming** the firmware verifies the status and finds it inconsistent since there is no fimware anymore. Something went wrong ...  so i think its OK to manually abort the previous upgrade.