libremesh/lime-packages #1132: Consider new interface names

ilario, 2024-09-24:
In the swconfig era (OpenWrt 19.07 and previous), interfaces had usually a name like eth0.2.
With DSA, interfaces can have funny names like [ethblue](https://github.com/openwrt/openwrt/blob/01996b785dcef61e6dadb50f61a13f905f3e497d/target/linux/ramips/dts/mt7621_gnubee_gb-pc2.dts#L148-L155), [game](
https://github.com/openwrt/openwrt/blob/eb13076e77c412cfe4dd6de48138507ea6f3c8b3/target/linux/mediatek/dts/mt7986a-acer-predator-w6.dts#L348-L351), lan1, ethernet1...

So we either find a general way to deal with arbitrary interface names, or we extract a list of the interface names from the targets we care about (for example, these: https://repo.libremesh.org/releases/2020.4-ow19/targets/).

The code that needs adaptation is the dev_parser in network.lua in lime-system:

https://github.com/libremesh/lime-packages/blob/master/packages/lime-system/files/usr/lib/lua/lime/network.lua#L228-L276

ilario, 2024-09-24:
Analyzing the targets that we usually use (), I found these strings that looks like interface names:

G10h4ck, 2024-09-24:
I hope at some point all, ethernet, wireless and usable devices which are embedded into the boeard, will be reported in board.json or in device tree, at that point we will not need anymore this filtering euristic, except maybe for pluggagle wifi or ethernet interfaces which usually doesn't have much complication like swconfig based switches (which we need to keep supporting as not all devices are DSA enabled)

pony1k, 2024-09-26:
I propose we extract interface names from /etc/board.json.

Every device that I encountered so far had their ethernet interfaces correctly listed in board.json. 

Here is a script that generates an approximate board.json for a given board_name: https://gist.github.com/pony1k/589cabfe3afa710adc8ec07d5ba2165a

All devices @ilario listed seem to have correct board.json:
```
./board-generate.sh ubnt,edgerouter-x-sfp
./board-generate.sh gnubee,gb-pc2
./board-generate.sh dlink,covr-x1860-a1
./board-generate.sh plasmacloud,pa2200
./board-generate.sh acer,predator-w6
./board-generate.sh dlink,covr-x1860-a1
./board-generate.sh zyxel,nr7101
./board-generate.sh ubnt,edgeswitch-8xp ath79/generic
./board-generate.sh qxwlan,e2600ac-c2
./board-generate.sh arcadyan,we420223-99
./board-generate.sh tplink,tl-wr941-v2 ath79/tiny
./board-generate.sh mqmaker,witi
```
@G10h4ck , do you have an example of a device where board.json is incorrect for ethernet interfaces?

G10h4ck, 2024-09-27:
As an example a device with an extra USB ethernet device plugged by the user.

ilario, 2024-11-07:
Amazing work @pony1k , thanks!
Regarding the USB ethernet, which would be the interface name? Does the current code manages it correctly?

ilario, 2024-12-07:
> Regarding the USB ethernet, which would be the interface name? Does the current code manages it correctly?

@G10h4ck if the board.json solution is better than the current one and has no drawbacks (the USB sticks are not supported currently) I would support the board.json approach.

The user experience when using a router with these interface names is terrible, see #1151.

So I would say we should fix this before making the release.

pony1k, 2024-12-08:
I'm only 99% sure, but I think the additional interfaces from USB dongles get names like eth0, eth1, etc. Those could be additionally collected to the ports found in board.json. Ideally, DSA conduit ports should be ignored. It would also be nice just in case to accept arbitrary names in port specific configurations, as long as they exist in /sys/class/net. So, if for whatever reason, there is a network interface called 'foo' that isn't listed in board.json, it would still be possible to configure it using `config net` and `option linux_name foo`.

PR #1154 (2025-01-22): Scrape device names from board.json
- Scrape configurable network devices from /etc/board.json.
- Avoid configuration on DSA conduit devices.
- Allow device specific configuration for network devices with
  arbitrary names. This is more intuitive from a users perpective. Before, devices whose names don't match certain patterns were ignored. This could be helpful for network devices with unforseen names.
- Remove 'nobridge' tag, as this seems no longer be needed.
- Add optional parameter 'port' to utils.is_dsa().
- Add 'dsa' tag for DSA user ports. This could be used to fi_x #1121 like suggeste [... cut]
closed 2025-01-22 by ilario