libremesh/lime-packages #1137: ubus call lime-utils get_node_status problem in some dsa routers

javierbrk, 2024-11-05:
a call to 
`ubus call lime-utils get_node_status`

wont always work

```
echo "" | /usr/libexec/rpcd/lime-utils call get_node_statusBusyBox v1.36.1 (2024-06-04 19:09:48 UTC) multi-call binary.

Usage: ip [OPTIONS] address|route|link|neigh|rule [ARGS]

OPTIONS := -f[amily] inet|inet6|link | -o[neline]

ip addr add|del IFADDR dev IFACE | show|flush [dev IFACE] [to PREFIX]
ip route list|flush|add|del|change|append|replace|test ROUTE
ip link set IFACE [up|down] [arp on|off] [multicast on|off]
 [promisc on|off] [mtu NUM] [name NAME] [qlen NUM] [address MAC]
 [master IFACE | nomaster] [netns PID]
ip neigh show|flush [to PREFIX] [dev DEV] [nud STATE]
ip rule [list] | add|del SELECTOR ACTION
**lua: /usr/lib/lua/lime/node_status.lua:141: attempt to index local 'dsa_json' (a nil value)**
```
the problem seems to be related to the use of unsupported options in the ip command on some specific ip compilations. The command that the scrip is trying to run is 

```bash

ip -j -p link show 
``` 
and some routers that use busybox don't have the -j option enabled.

there must be a way to pragmatically include a "full" ip package or change the code to use the simple ip and some tools to parse the output.

ilario, 2024-11-05:
Does this affect also the  2024.1-rc1 release candidate?
Can you post the output of that command with the full IP package and the most similar output obtainable with the default IP shipped with OpenWrt?

ilario, 2024-11-05:
This command was added by @selankon in https://github.com/libremesh/lime-packages/commit/162925a2281d2bf5e5a3671de9eb129a0acec024

By default, OpenWrt has ip-tiny instead of ip-full. Likely we did not realize this before as lime-debug depends on "ip", that I suppose pulls "ip-full"...?

The differences between the two versions are mainly these ones.
https://github.com/openwrt/openwrt/blob/openwrt-23.05/package/network/utils/iproute2/patches/170-ip_tiny.patch
It does not mention anything about JSON... Mah... (seems that I missed something)

javierbrk, 2024-11-06:
> Can you post the output of that command with the full IP package and the most similar output obtainable with the default IP shipped with OpenWrt?

I tried with the routers on my desk but I don't have enough space, not even for ip_tiny. This will take some more time. 

> Does this affect also the 2024.1-rc1 release candidate?
i'm not sure

ilario, 2024-11-07:
> Does this affect also the 2024.1-rc1 release candidate?

Looks like it does affect also the release candidate:
https://github.com/libremesh/lime-packages/blob/v2024.1-rc1/packages/ubus-lime-utils/files/usr/lib/lua/lime/node_status.lua#L125

javierbrk, 2024-11-07:
> I mean: can you post the output of the ip command that you have on your router?

yess ! 

```
BusyBox v1.36.1 (2024-06-04 19:09:48 UTC) multi-call binary.

Usage: ip [OPTIONS] address|route|link|neigh|rule [ARGS]

OPTIONS := -f[amily] inet|inet6|link | -o[neline]

ip addr add|del IFADDR dev IFACE | show|flush [dev IFACE] [to PREFIX]
ip route list|flush|add|del|change|append|replace|test ROUTE
ip link set IFACE [up|down] [arp on|off] [multicast on|off]
 [promisc on|off] [mtu NUM] [name NAME] [qlen NUM] [address MAC]
 [master IFACE | nomaster] [netns PID]
ip neigh show|flush [to PREFIX] [dev DEV] [nud STATE]
ip rule [list] | add|del SELECTOR ACTION
```

The image I was using is self compiled. It uses the ip command available from busybox. 

Using the precompiled libremesh image 2024.1-rc1-ow23.05.3-default witch comes with "ip-tiny - 6.3.0-1" there is no problem. 
The command will behave as expected. 
```

root@LiMe-fc3abd:~# ip -j -d link show eth0
[{"ifindex":2,"ifname":"eth0","flags":["BROADCAST","MULTICAST","UP","LOWER_UP"],"mtu":1500,"qdisc":"fq_codel","operstate":"UP","linkmode":"DEFAULT","group":"default","txqlen":1000,"link_type":"ether","address":"c0:4a:00:fc:3a:bd","broadcast":"ff:ff:ff:ff:ff:ff","promiscuity":2,"min_mtu":68,"max_mtu":16359,"inet6_addr_gen_mode":"eui64","num_tx_queues":1,"gso_max_size":65536,"gso_max_segs":65535,"parentbus":"platform","parentdev":"1a000000.eth"}]
```

using plain OpenWrt 23.05.5, r24106-10cc5fcd00
 ```

root@OpenWrt:~# ip
BusyBox v1.36.1 (2024-09-23 12:34:46 UTC) multi-call binary.

Usage: ip [OPTIONS] address|route|link|neigh|rule [ARGS]

OPTIONS := -f[amily] inet|inet6|link | -o[neline]

ip addr add|del IFADDR dev IFACE | show|flush [dev IFACE] [to PREFIX]
ip route list|flush|add|del|change|append|replace|test ROUTE
ip link set IFACE [up|down] [arp on|off] [multicast on|off]
	[promisc on|off] [mtu NUM] [name NAME] [qlen NUM] [address MAC]
	[master IFACE | nomaster] [netns PID]
ip neigh show|flush [to PREFIX] [dev DEV] [nud STATE]
ip rule [list] | add|del SELECTOR ACTION
root@OpenWrt:~# 
```
I think that a simple solution it wold be to add ip-tiny to the package dependencies. I can also create a pull with a proposal that only uses the plain command without -j option and parses the output using string comparison.

ilario, 2024-11-07:
Thanks!
The current code looks for the following info: ifindex (2 in your example), link (absent), ifname (eth0), operstate (UP).
Surely it is possible to get this data without ip-tiny, but that package is just 140 kB and I think we can just add it as a dependency. Adding `ip` as a dependency should be enough.

closed 2024-12-27 by G10h4ck