libremesh/lime-packages #1138: option htmode 'HT20' create HT40 mesh interfaces.

der-nele, 2024-11-08:
setup
OpenWrt 23.05.5, r24106-10cc5fcd00
with BMX7 and batman

i have problem with the HT mode
i choose HT 20. in the lime node
and some interfaces use HT 20. but in the mesh interfaces there are some HT 40 apply 
wold be nice if some one can help me.

[image]
[image]

der-nele, 2024-11-08:
do not work in the way i try
root@kh-bad:~# lime-report
-ash: lime-report: not found

der-nele, 2024-11-08:
our packet list is: 
-dnsmasq -odhcpd-ipv6only

-batctl-default batctl-full

lime-system
lime-proto-anygw
lime-proto-bmx7
lime-proto-batadv
shared-state-bat_hosts
shared-state-dnsmasq_hosts
shared-state-dnsmasq_leases
shared-state-nodes_and_links
shared-state-network_nodes

bmx7-topology
bmx7-iwinfo

-ppp-mod-pppoe -ppp -kmod-ppp -kmod-pppoe -kmod-pppox
-kmod-usb-ledtrig-usbport -kmod-usb-net-qmi-wwan -kmod-usb-ohci -kmod-usb-serial-option -kmod-usb2

-luci-base -luci-compat -luci-lib-base

wifi-unstuck-wa
-wpad-basic-mbedtls wpad-mbedtls
ca-certificates ca-bundle rpcd-mod-iwinfo

ilario, 2024-11-08:
> do not work in the way i try root@kh-bad:~# lime-report -ash: lime-report: not found

Then install it with:

opkg install lime-report

or click on the link I included in my comment above and copy and paste the lime-report file content to a file in the router. Then make it executable with:

chmod a+x lime-report

and execute it with:

./lime-report > /tmp/report

and upload that file inside a comment here (I suppose you can upload it the same way you add a picture. Otherwise copy it on a pastevin service online.

der-nele, 2024-11-10:
[report.txt](https://github.com/user-attachments/files/17690881/report.txt)

der-nele, 2024-11-10:
pony had also a look on it and say it is possible that the scanning tool i use show wrong information about it.

ilario, 2024-11-10:
Checking the content of lime-report, it seems that all interfaces have HT20.
You can see that running the `iwinfo` command.

I agree with Pony, it could be that your detection software is failing. Or there is some weird thing happening in OpenWrt. Please try measuring the channel width with another software.

der-nele, 2024-11-10:
i do not finde a scanner with is showing the hidden mesh interfaces and the hw mode.

ilario, 2024-11-10:
Then I can see no way to figure this out.
Shall we close this issue, supposing that the software was wrong?

der-nele, 2024-11-11:
yes for me is fine to close. i will open again if i find a reason fore.
thanks

closed 2024-11-11 by ilario