libremesh/lime-packages #1146: Setting a static IP on the WAN port

ilario, 2024-11-24:
I just tried to set a static IPv4 on the WAN port connected to an internet-providing router and just realized that Babeld does not share the default route from this interface.

The configuration I inserted in `/etc/config/lime-node` is this:

```
config net lm_hwd_openwrt_wan
	option autogenerated 'false'

config net wan_static
	option linux_name 'wan'
	list protocols 'static'
	option static_ipv4 '192.168.1.2/24'
	option static_gateway_ipv4 '192.168.1.1'
```

(as documented in [lime-example.txt](https://github.com/libremesh/lime-packages/blob/master/packages/lime-docs/files/www/docs/lime-example.txt), the first section is for avoiding the [lime-hwd-openwrt-wan](https://github.com/libremesh/lime-packages/tree/master/packages/lime-hwd-openwrt-wan) to configure the wan interface to use dhcp client, the second is creating a section setting the static IP).

The default route is present, but it does not get shared with the rest of the network.

I suppose this happens because [babeld-auto-gw-mode](https://github.com/libremesh/lime-packages/tree/master/packages/babeld-auto-gw-mode) is looking for the default route only on the section `wan` of `/etc/config/network`:

https://github.com/libremesh/lime-packages/blob/1aa97c63ab77c94b9ba905f7f6a91dd0143bc990/packages/babeld-auto-gw-mode/files/etc/watchping/wan-ok.d/babeld-gw#L14

but the configuration above generates the route in a section named `lm_net_wan_static`:

```
config interface 'wan'
	option device 'wan'
	option proto 'none'

config interface 'wan6'
	option device 'wan'
	option proto 'none'

[...]

config interface 'lm_net_wan_static'
	option proto 'static'
	option auto '1'
	option ifname 'wan'
	option ipaddr '192.168.1.2'
	option netmask '255.255.255.0'
	option gateway '192.168.1.1'
```

a-gave, 2024-11-26:
Hi, add few notes and a question on this.
This reminds me of a gateway where i left dhcp as proto for wan.
Testing on that gateway and replacing `network.interface.wan` with `lm_net_eth0_2_static` (in my case) 
in `/etc/watchping/wan-{ok,fail}.d/babeld-gw` allows to use the static ip:
```
local wan4_status = conn:call("network.interface.lm_net_eth0_2_static", "status", {})
```
To avoid this manual change i tried to rewrite both scripts to not use `ubus` 
but to look only at `ip route` as in the scripts copied below, for retrieving the correct `route` and `device`.

But testing this i noticed that watchping will check only `wan`, `wan6` and `client-wwan` and wont check i.e. `lm_net_wan_static`
... Hopefully in this case it works anyway because it pings on device `eth0.2` 
as it takes `ifname` from the existing interface `wan`  that use the same device as `lm_new_wan_static`

As a proposal, could have sense instead to define an overwriteable array of main wan interfaces? And let both watchping and babeld-auto-gw-mode check this list to do their stuff? 
For example in uci:
```
uci set lime-defaults.main_wan_interfaces='wan wan6 client-wwan lm_net_wan_static lm_net_wan6_static'
```
allowing so, to do this in my case:
```
uci set lime-node.main_wan_interfaces='lm_net_eth0_2_static'
```
babeld-auto-gw-mode with or without ubus doesn't matter at this point, but i wonder: 
- what would be the best way to handle metrics codes ('7' '84831') for multiple wan ifaces in babeld watchping's scripts?
- should it set `proto 7` all default routes, using maybe different metrics starting from 10, 
or should it only set `proto 7` on the best (as it is done in the scripts below that check only first default route with metric 0)?

Both watchping and babeld-auto-gw-mode should in this case `create the dir` or `copy the scripts` to /etc/watchping/$interface-$reason.d/ taking the $interface from this list

Here the example of babel watchping's hooks without ubus that check `route` and `device` from iproute:
<details>
<summary>/etc/watchping/wan-fail.d/babeld-gw</summary>
<pre><code>
#!/usr/bin/lua
--! Copyright (C) 2021 Santiago Piccinini <spiccinini@altermundi.net>
--!
--! This is free software, licensed under the GNU Affero General Public License v3.
local utils = require "lime.utils"
local NONWORKING_ROUTE_METRIC = 84831
local wan4_dev = utils.unsafe_shell("ip r s default metric 0 | cut -d ' ' -f 5"):gsub("\n","")

if wan4_dev ~= '' then 
    --! Do our best to select only the "original" default route installed. Using grep because
    --! using the dev argument removes the dev part of the ip route output.
    local route = utils.unsafe_shell("ip r s default metric 0 | grep " .. wan4_dev):gsub("\n","")

    --! replace the default route with one that has metric NONWORKING_ROUTE_METRIC
    if route ~= '' then
        os.execute("ip r add " .. route .. " metric " .. tonumber(NONWORKING_ROUTE_METRIC))
        os.execute("ip r del " .. route)
    end

    --! remove the babel redistributable route
    os.execute("ip r del default proto 7")
end
</code></pre>
</details>

<details>
<summary>/etc/watchping/wan-ok.d/babeld-gw</summary>
<pre><code>
#!/usr/bin/lua
--! Copyright (C) 2021 Santiago Piccinini <spiccinini@altermundi.net>
--!
--! This is free software, licensed under the GNU Affero General Public License v3.
local utils = require "lime.utils"
local wan4_dev_not = utils.unsafe_shell("ip r s default metric 84831 | cut -d ' ' -f 5"):gsub("\n","")

if wan4_dev_not ~= '' then 
    --! change the default route from the nonworking metric value to the working metric value
    local nonworking_route = utils.unsafe_shell("ip r s default metric 84831 | grep " .. wan4_dev_not):gsub("\n","")
    if nonworking_route ~= '' then
        os.execute("ip r add " .. nonworking_route .. " metric 0")
        os.execute("ip r del " .. nonworking_route .. " metric 84831")
    end
end

local wan4_dev = utils.unsafe_shell("ip r s default metric 0"):gsub("\n","")

if wan4_dev ~= '' then
    --! install a route with proto value 7 that will be redistributed by babeld
    local r = utils.unsafe_shell("echo " .. wan4_dev .. " | sed 's|\\(.*\\) proto.*|ip r a \\1 proto 7 metric 10|'"):gsub("\n","")
    os.execute(r)
end
</code></pre>
</details>

ilario, 2024-11-26:
An alternative could be to add the IP and gateway IP as options of the wan proto.

Either something like this:

```
config net lm_hwd_openwrt_wan
	option autogenerated 'false'
	option linux_name 'wan'
	list protocols 'wan:192.168.1.2/24:192.168.1.1'
```

or something like this:

```
config net lm_hwd_openwrt_wan
	option autogenerated 'false'
	option linux_name 'wan'
	list protocols 'wan'
	option wan_static_ipv4 '192.168.1.2/24'
	option wan_static_gateway_ipv4 '192.168.1.1'
```

These IPs will then be taken and used for setting the static configuration here: https://github.com/libremesh/lime-packages/blob/master/packages/lime-proto-wan/files/usr/lib/lua/lime/proto/wan.lua#L16

We can take some code from the static proto:
https://github.com/libremesh/lime-packages/blob/master/packages/lime-system/files/usr/lib/lua/lime/proto/static.lua

Jadhav-Dhanashri, 2026-01-17:
Hello maintainers ,
I'm new to freifunk.net open source contribution & this would be my first issue. I'd like to work on this if it's available & learn from the process. Could you please assign it to me?
Thank you !

ilario, 2026-01-17:
> Hello maintainers , I'm new to freifunk.net open source contribution & this would be my first issue. I'd like to work on this if it's available & learn from the process. Could you please assign it to me? Thank you !

I did forget this issue! Thanks @Jadhav-Dhanashri for bringing this up in my mind.
Reading again my proposals (`wan:192.168.1.2/24:192.168.1.1` and `wan_static_ipv4`) I now find them horrid. So it would be much better to improve the code of babeld-auto-gw-mode and watchping as proposed by @a-gave.
@Jadhav-Dhanashri if you want to go on completing @a-gave implementation, please feel free!