libremesh/lime-packages #1150: Error messages from shared-state-async

ilario, 2024-11-30:
Testing the code from the branch 2024.1, I observed a lot of messages from shared-state-async in logread.

Here I paste the whole output of logread from the three routers composing the network, hoping that they can be useful for shared-state-async development (or reducing its verbosity if they are not error messages).

[logread_router1.txt](https://github.com/user-attachments/files/17965889/logread_router1.txt)

[logread_router2.txt](https://github.com/user-attachments/files/17965890/logread_router2.txt)

[logread_router3.txt](https://github.com/user-attachments/files/17965891/logread_router3.txt)

ilario, 2025-01-11:
Duplicate of #1081 and already fixed in #1082

ilario, 2025-01-11:
Not a duplicate, sorry.
The problem is way less severe than in #1081 but there are still too many log messages.

G10h4ck, 2025-01-12:
Messages like 
`daemon.err shared-state-async[1214]: W 1732977611.376 std::task<int> SharedState::merge(...) Discarding received known entry: aa:bb:cc:dd:ee:ff authored by this node with higher TTL from remote peer: ipv6://[fe80::2841:ddff:bbcc:ddee%25br-lan]:3490 is remote peer ill?`
Are signaling a really problematic situation happening in the network so should not be suppressed, one should instead debug what's happening. It seems the other node could have not bleached his view of the state in time. Is the other node overloaded?

while messages like
`daemon.err shared-state-async[1214]: D 1732977611.467 std::task<int> SharedState::merge(...) bat-hosts got 0 significative changes out of 29 input slice size: 33 state size: 33` can be suppressed in a production build

ilario, 2025-01-12:
> while messages like `daemon.err shared-state-async[1214]: D 1732977611.467 std::task<int> SharedState::merge(...) bat-hosts got 0 significative changes out of 29 input slice size: 33 state size: 33` can be suppressed in a production build

Please suppress them, they are filling the logs so that more important log messages are being erased. It would be great to not having them in the upcoming release.

ilario, 2025-01-17:
In a simple network with no load at all (just a ping) running the upcoming release 2024.1 I have these messages in just a couple of minutes:

### TP-Link WDR3600

```
# logread -f
Fri Jan 17 22:20:58 2025 daemon.err shared-state-async[1480]: D 1737152458.793 ReturnType AwaitableSyscall<SyscallOp, ReturnType, multiShot, errorValue>::await_resume() [with SyscallOp = ConnectOperation; ReturnType = int; bool multiShot = false; ReturnType errorValue = -1] syscall failed on resume mReturnValue: -1 error: 146 Connection refused category: generic
Fri Jan 17 22:20:58 2025 daemon.err shared-state-async[1480]: D 1737152458.798 ReturnType AwaitableSyscall<SyscallOp, ReturnType, multiShot, errorValue>::await_resume() [with SyscallOp = ConnectOperation; ReturnType = int; bool multiShot = false; ReturnType errorValue = -1] syscall failed on resume mReturnValue: -1 error: 146 Connection refused category: generic
Fri Jan 17 22:20:58 2025 daemon.err shared-state-async[1480]: D 1737152458.801 ReturnType AwaitableSyscall<SyscallOp, ReturnType, multiShot, errorValue>::await_resume() [with SyscallOp = ConnectOperation; ReturnType = int; bool multiShot = false; ReturnType errorValue = -1] syscall failed on resume mReturnValue: -1 error: 146 Connection refused category: generic
Fri Jan 17 22:20:58 2025 daemon.err shared-state-async[1480]: D 1737152458.805 ReturnType AwaitableSyscall<SyscallOp, ReturnType, multiShot, errorValue>::await_resume() [with SyscallOp = ConnectOperation; ReturnType = int; bool multiShot = false; ReturnType errorValue = -1] syscall failed on resume mReturnValue: -1 error: 146 Connection refused category: generic
Fri Jan 17 22:21:00 2025 cron.err crond[1540]: USER root pid 13896 cmd ((sleep $((RANDOM % 120)); shared-state-async insert net-stats < /tmp/shared-state/network_statistics.json &> /dev/null)&)
Fri Jan 17 22:21:09 2025 daemon.err shared-state-async[1480]: D 1737152469.323 std::task<int> SharedState::merge(const std::string&, const std::map<std::__cxx11::basic_string<char>, StateEntry>&, const sockaddr_storage&, std::error_condition*) net-stats got 1 significative changes out of 1 input slice size: 1 state size: 1
Fri Jan 17 22:21:09 2025 daemon.err shared-state-async[1480]: D 1737152469.384 std::task<bool> SharedState::notifyHooks(const std::string&, std::error_condition*) net-stats
Fri Jan 17 22:21:28 2025 daemon.err shared-state-async[1480]: D 1737152488.854 ReturnType AwaitableSyscall<SyscallOp, ReturnType, multiShot, errorValue>::await_resume() [with SyscallOp = ConnectOperation; ReturnType = int; bool multiShot = false; ReturnType errorValue = -1] syscall failed on resume mReturnValue: -1 error: 146 Connection refused category: generic
Fri Jan 17 22:21:28 2025 daemon.err shared-state-async[1480]: D 1737152488.859 ReturnType AwaitableSyscall<SyscallOp, ReturnType, multiShot, errorValue>::await_resume() [with SyscallOp = ConnectOperation; ReturnType = int; bool multiShot = false; ReturnType errorValue = -1] syscall failed on resume mReturnValue: -1 error: 146 Connection refused category: generic
Fri Jan 17 22:22:00 2025 cron.err crond[1540]: USER root pid 14211 cmd ((for dataFile in /var/shared-state/data/* ; do shared-state bleach $(basename $dataFile .json); done &> /dev/null)&)
Fri Jan 17 22:22:01 2025 daemon.err shared-state-async[1480]: D 1737152521.761 ReturnType AwaitableSyscall<SyscallOp, ReturnType, multiShot, errorValue>::await_resume() [with SyscallOp = ConnectOperation; ReturnType = int; bool multiShot = false; ReturnType errorValue = -1] syscall failed on resume mReturnValue: -1 error: 146 Connection refused category: generic
Fri Jan 17 22:22:01 2025 daemon.err shared-state-async[1480]: D 1737152521.765 ReturnType AwaitableSyscall<SyscallOp, ReturnType, multiShot, errorValue>::await_resume() [with SyscallOp = ConnectOperation; ReturnType = int; bool multiShot = false; ReturnType errorValue = -1] syscall failed on resume mReturnValue: -1 error: 146 Connection refused category: generic
Fri Jan 17 22:22:28 2025 daemon.err shared-state-async[1480]: D 1737152548.813 ReturnType AwaitableSyscall<SyscallOp, ReturnType, multiShot, errorValue>::await_resume() [with SyscallOp = ConnectOperation; ReturnType = int; bool multiShot = false; ReturnType errorValue = -1] syscall failed on resume mReturnValue: -1 error: 146 Connection refused category: generic
Fri Jan 17 22:22:28 2025 daemon.err shared-state-async[1480]: D 1737152548.817 ReturnType AwaitableSyscall<SyscallOp, ReturnType, multiShot, errorValue>::await_resume() [with SyscallOp = ConnectOperation; ReturnType = int; bool multiShot = false; ReturnType errorValue = -1] syscall failed on resume mReturnValue: -1 error: 146 Connection refused category: generic
Fri Jan 17 22:23:00 2025 daemon.err shared-state-async[1480]: D 1737152580.772 ReturnType AwaitableSyscall<SyscallOp, ReturnType, multiShot, errorValue>::await_resume() [with SyscallOp = ConnectOperation; ReturnType = int; bool multiShot = false; ReturnType errorValue = -1] syscall failed on resume mReturnValue: -1 error: 146 Connection refused category: generic
Fri Jan 17 22:23:00 2025 daemon.err shared-state-async[1480]: D 1737152580.777 ReturnType AwaitableSyscall<SyscallOp, ReturnType, multiShot, errorValue>::await_resume() [with SyscallOp = ConnectOperation; ReturnType = int; bool multiShot = false; ReturnType errorValue = -1] syscall failed on resume mReturnValue: -1 error: 146 Connection refused category: generic
[... cut ...]
Fri Jan 17 22:23:41 2025 daemon.err shared-state-async[1290]: D 1737152621.805 std::task<int> SharedState::merge(const std::string&, const std::map<std::__cxx11::basic_string<char>, StateEntry>&, const sockaddr_storage&, std::error_condition*) net-stats got 0 significative changes out of 0 input slice size: 1 state size: 1
Fri Jan 17 22:23:42 2025 daemon.err babeld[1662]: Interface ethernet1_17 has no link-local address.
Fri Jan 17 22:23:42 2025 daemon.err babeld[1662]: setsockopt(IPV6_LEAVE_GROUP): Address not available
```