libremesh/lime-packages #1162: Running an x86/64 image with tools/qemu_dev_start returns "eth0 doesn't exists!"

a-gave, 2025-02-04:
It's probably not widely used, but it was working until recently
And it is not working properly with current last HEAD: ca63283
```
 ./qemu_dev_start --node-id 1 --enable-wan wlo1 rootfs.tar.gz kernel.bin
```
These are the logs encountered just after the boot:
```
root@OpenWrt:/# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
root@OpenWrt:/# lime-config 
Cannot acquire lock on /var/run/lime-config.pid
This may be caused by lime-config already running or unexpectedly crashed.
If it is the latter case (check carefully!) try by deleting the lock file with:
# rm -rf /var/run/lime-config.pid
Before running lime-config again
root@OpenWrt:/# rm -rf /var/run/lime-config.pid
root@OpenWrt:/# lime-config 
/usr/bin/lua: /usr/lib/lua/lime/network.lua:55: network.primary_interface() eth0 doesn't exists!
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/lime/network.lua:55: in function 'assert_interface_exists'
	/usr/lib/lua/lime/network.lua:65: in function 'primary_interface'
	/usr/lib/lua/lime/network.lua:70: in function 'primary_mac'
	/usr/lib/lua/lime/config.lua:190: in function 'main'
	/usr/bin/lime-config:55: in main chunk
	[C]: ?
root@OpenWrt:/#
```

Instead the ext4-combined-efi image setup correctly all interfaces, tested running:
```
gunzip libremesh-master-ow23.05.5-ca63283-20250203-default-x86-64-generic-ext4-combined-efi.img.gz
mv libremesh-master-ow23.05.5-ca63283-20250203-default-x86-64-generic-ext4-combined-efi.img combined-efi.img
qemu-system-x86_64 -M q35 -drive file=combined-efi.img,id=d0,if=none,bus=0,unit=0 -device ide-hd,drive=d0,bus=ide.0

javierbrk, 2025-04-23:
I confirm the issue. I was trying to figure out what is going on... this is what I got so far. 

with ./qemu_dev_start libremesh-2024.1-ow23.05.5-default-x86-64-rootfs.tar.gz libremesh-2024.1-ow23.05.5-default-x86-64-generic-kernel.bin these files I got the machine up and running but it seems that is not able to rise any ethernet networks, tried loading the driver with insmod but it fails...
```
root@OpenWrt:/#insmod e1000
Failed to find e1000. Maybe it is a built in module ?
root@OpenWrt:/# 
root@OpenWrt:/# ls /lib/modules/$(uname -r)
batman-adv.ko           nf_flow_table_inet.ko   nft_log.ko
button-hotplug.ko       nf_flow_table_ipv4.ko   nft_masq.ko
cfg80211.ko             nf_flow_table_ipv6.ko   nft_meta_bridge.ko
compat.ko               nf_log_syslog.ko        nft_nat.ko
crc-ccitt.ko            nf_nat.ko               nft_numgen.ko
crypto_acompress.ko     nf_reject_ipv4.ko       nft_objref.ko
fat.ko                  nf_reject_ipv6.ko       nft_quota.ko
libcrc32c.ko            nf_tables.ko            nft_redir.ko
lzo-rle.ko              nfnetlink.ko            nft_reject.ko
lzo.ko                  nft_chain_nat.ko        nft_reject_bridge.ko
lzo_compress.ko         nft_counter.ko          nft_reject_inet.ko
lzo_decompress.ko       nft_ct.ko               nft_reject_ipv4.ko
macvlan.ko              nft_fib.ko              nft_reject_ipv6.ko
md5.ko                  nft_fib_inet.ko         nls_cp437.ko
nf_conntrack.ko         nft_fib_ipv4.ko         nls_iso8859-1.ko
nf_conntrack_bridge.ko  nft_fib_ipv6.ko         nls_utf8.ko
nf_defrag_ipv4.ko       nft_flow_offload.ko     vfat.ko
nf_defrag_ipv6.ko       nft_hash.ko
nf_flow_table.ko        nft_limit.ko
root@OpenWrt:/# 
```
but in the combined image the driver is present and interfaces can be loaded. 

Why the combined image has the driver and the other doesn't? can we customize the content of one of them ? the json that describes the build includes the driver as a module https://downloads.libremesh.org/releases/2024.1-ow23.05.5/targets/x86/64/default/profiles.json

javierbrk, 2025-04-23:
There is a reported similar issue in openwrt https://github.com/openwrt/openwrt/issues/11730 

The last images that work are 

https://downloads.libremesh.org/releases/2024.1-rc1-ow23.05.3-default/targets/x86/64/libremesh-2024.1-rc1-ow23.05.3-default-x86-64-generic-rootfs.tar.gz

https://downloads.libremesh.org/releases/2024.1-rc1-ow23.05.3-default/targets/x86/64/libremesh-2024.1-rc1-ow23.05.3-default-x86-64-generic-initramfs-kernel.bin

There must be some config where the vital modules are listed and that list changed.

a-gave, 2025-04-23:
Tested with latest commit [ca63283](https://github.com/libremesh/lime-packages/commit/ca6328347b97728b796fa28d243e02f95f7c6616) against openwrt-24.10.1 and works as expected