libremesh/lime-packages #1163: unsync docs

javierbrk, 2025-02-27:
This commit https://github.com/libremesh/lime-packages/commit/6c9a20a2861f2e34ff0f82d63c99c06215f826d9 changes package name, but the docs still use package name with the typo. https://libremesh.org/packages/deferable-reboot.html

ilario, 2025-02-27:
Yep, @a-gave created those pages using some black magic and he [updated them in July 2024](https://github.com/libremesh/libremesh.github.io/commit/ab9115285c1d890c5367d8f5e29f696d16074854).

The only Makefiles that changed since July are:

[packages/shared-state-babel_links_info/Makefile](https://github.com/libremesh/lime-packages/pull/1112/files#diff-bdf9ac048e493659e7512cfae4aabdc0afc024fe85c90206e273695e170bb737)
[packages/shared-state-wifi_links_info/Makefile](https://github.com/libremesh/lime-packages/commit/f46ea9aa5c8f14ee5520b3bb5d92607923906d3c#diff-512766a52548ac9d3890d1bf72d129dd44c8499cff88cefb3cd88ec5f5bbd3d0)
[packages/lime-app/Makefile](https://github.com/libremesh/lime-packages/commit/4dcf970215d64976e050cd458d40ff61bf1e4e76#diff-4794fa52d938ecb122d9abbbaf3386632dc96010c26034832e1c273a3b086330)
[packages/ubus-lime-groundrouting/Makefile](https://github.com/libremesh/lime-packages/commit/16cb12e71ad3a8bbbbb0db0214faf69c6b48c5fb#diff-cf56ff69206b22ec2aa75b0e1af25704694b74873fc6406c51f6cad5e76b80ab)

and, as you mentioned, also:

packages/deferrable-reboot/Makefile

So, as they are few, maybe the fastest solution is to copy and paste them individually...
@javierbrk pull requests are welcome XD

a-gave, 2025-07-02:
Hi this was closed in https://github.com/libremesh/libremesh.github.io/commit/8f1af7337312670dd67d8690707571ddf7340b10

Minor notes to myself:
- remove the broken link http://libremesh.org/packages/packages_list.html
- Homogenize readme names into README.adoc or REAME.md

closed 2025-07-02 by a-gave