libremesh/lime-packages #1169: unable to open LuCI, unable to compile as Lua template

maciejopalinski, 2025-04-08:
Device: TP-Link Archer D7 v1
Firmware: LiMe master development (master rev. ca63283 20250123_1743)

How to reproduce:
1. Visit https://thisnode.info/cgi-bin/luci

```
Unable to render any theme header template, last error was: Unable to compile 'themes/bootstrap-light/header' as Lua template:

/usr/lib/lua/luci/ucodebridge.lua:23: /usr/lib/lua/luci/template.lua:158: Failed to load template 'themes/bootstrap-light/header'. Error while parsing template '/usr/lib/lua/luci/view/themes/bootstrap-light/header.htm': No such file or directory In error(), file [C] called from function run (/usr/lib/lua/luci/ucodebridge.lua:23) called from function [anonymous function] (/usr/lib/lua/luci/ucodebridge.lua:31)
In [anonymous function](), file /usr/share/ucode/luci/runtime.uc, line 106, byte 62:
  called from function [anonymous function] (/usr/share/ucode/luci/runtime.uc:166:60)
  called from function [anonymous function] (/usr/share/ucode/luci/dispatcher.uc:895:3)
  called from anonymous function (/www/cgi-bin/luci:39:13)

 `                    vm.get('_G', 'luci', 'ucodebridge', 'compile').call(path);`
  Near here ------------------------------------------------------------------^
```

ilario, 2025-04-09:
Hi!
Thanks for the report!
Is LuCI installed?

We did not include LuCI in the [list of reco](https://libremesh.org/development.html#compiling_libremesh_from_source_code)m[mended packages](https://github.com/libremesh/network-profiles/blob/master/libremesh/suggested-packages/Makefile) as:
1) there is lime-app that includes some of LuCI's features
2) LuCI edits directly the configuration files, and these edits could be overwritten if then you issue the `lime-config` command (the recommended way to configure a node is editing the /etc/config/lime-node file following [this documentation](https://github.com/libremesh/lime-packages/blob/master/packages/lime-docs/files/www/docs/lime-example.txt) and then to launch the `lime-config` command).

Clearly, you can install LuCI and use it.
I am not sure of the package name, but you can try with something like this:

```
opkg update
opkg install luci
```

Let us know if it works!

maciejopalinski, 2025-04-10:
Oh! I didn't know that luci wasn't installed. I built the image with only the recommended packages using exactly the guide that you linked. I will make sure to install luci as well. I will let you know if it works! Thank you

maciejopalinski, 2025-04-10:
It works after running `opkg install luci`