libremesh/lime-packages #1174: luci-mod-status remove it or rename it?

a-gave, 2025-05-22:
libremesh: 2024.1
openwrt: 24.10.1

the package `luci-mod-status` duplicate the official one from `openwrt/luci` feed and prevent installing `luci` alongside lime-packages with imagebuilder because of a missing dependency on `luci-lib-jquery-1-4`

```
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci:
 * 	luci-lib-jquery-1-4
 * opkg_install_cmd: Cannot install package luci.
make[2]: *** [Makefile:234: package_install] Error 255
make[1]: *** [Makefile:161: _call_image] Error 2
make: *** [Makefile:337: image] Error 2
```

```
root@ninux-e63db5:~# opkg list | grep luci-mod-status
luci-mod-status - 25.137.37373~691440a - LuCI Status Pages
luci-mod-status - 2024-12-10-1733846299 - Minimal and simple web app to view some status info: * a realtime plot of wifi neighbours signal (think "iw station dump") * batman-adv mesh nodes (if luci-app-batman-adv is installed) * bmx6 mesh nodes (if luci-app-bmx6 is installed) * ethernet switch ports status
```

ilario, 2025-05-23:
Let's delete it!

It is in the list of the to-be-removed packages: https://github.com/libremesh/lime-packages/issues/977#issuecomment-2024963011

Could you make a pull request with its removal? Thanks!!

a-gave, 2025-07-02:
Closed by #1176