libremesh/lime-packages #1177: Replacing deferrable-reboot with watchcat

AguTrachta, 2025-05-29:
## Objective

We want to allow that watchcat configuration can be made inside LiMe Packages, 

The initial idea is to create a new package that includes OpenWrt's watchcat as a dependence like `lime-watchcat`,  with its own config file:

```
config watchcat
        option enabled '1'
	option period '6h'
	option mode 'ping_reboot'
	option pinghosts '8.8.8.8'
	option forcedelay '30'
```

Also we can implement an integration script that reads the config file and apply those values on the real watchcat

ilario, 2025-05-30:
The easy way, in my opinion, to have the `watchcat` configuration updated at each execution of `lime-config`, would be to create a `lime-hwd-watchcat` package, similar to the [lime-hwd-ground-routing](https://github.com/libremesh/lime-packages/tree/master/packages/lime-hwd-ground-routing) one.

@G10h4ck what do you think?

ilario, 2025-06-03:
> one specific mode as a default replacement for deferrable-reboot (like the periodic reboot mode), or should I provide a way to choose and configure any of these modes?

That's a good question!
In my opinion, supporting the configuration of Ping Reboot is enough.
Ping Reboot is what deferrable-reboot does, no?
Still in my opinion, there's no need to spend time supporting every possible option, and that's not what we usually do (e.g. with lime-proto-batadv we add support for options we consider are useful to change in LibreMesh networks, not for all of the possible settings).

ilario, 2025-07-31:
Fixed with #1186

closed 2025-07-31 by ilario