libremesh/lime-packages #1178: Adding Wi-Fi Support to QEMU Simulations in LibreMesh

javierbrk, 2025-06-04:


VGDSpehar, 2025-08-27:
You can find on this repository : 
https://github.com/VGDSpehar/libremesh-virtual-mesh

Instructions and scripts for setting up a Wi-Fi mesh of 3 LibreMesh Node, in a virtualized environment using :
- QEMU
- mac80211_hwsim
- vwifi for relaying Wi-Fi frames across guests

ilario, 2026-09-05:
@javierbrk can you check which ones of the sub-issues are still relevant and should be kept open?

ilario, 2026-09-05:
In general, for completing this I believe we should merge the GSoC work from @VGDSpehar that can be found here https://github.com/VGDSpehar/libremesh-virtual-mesh, their code should go to the `tools/` folder.
The instructions from the readme of that repository should expand the `VIRTUALIZING.md` file.

Also, it would be very useful to have the `vwifi` package https://github.com/javierbrk/vwifi_cli_package included in OpenWrt's repositories. @javierbrk can you propose them the package for inclusion?