libremesh/lime-packages #1190: [2024.1-ow24.10.1] RuntimeError: Impossible package selection

VGDSpehar, 2025-06-11:
When requesting a custom firmware build in the LibreMesh firmware selector for 
2024.1-ow24.10.1-mini with `lime-app` included in the packages, i get this error : 

`Traceback (most recent call last): File "/usr/local/lib/python3.12/site-packages/rq/worker.py", line 1633, in perform_job return_value = job.perform() ^^^^^^^^^^^^^ File "/usr/local/lib/python3.12/site-packages/rq/job.py", line 1331, in perform self._result = self._execute() ^^^^^^^^^^^^^^^ File "/usr/local/lib/python3.12/site-packages/rq/job.py", line 1365, in _execute result = self.func(*self.args, **self.kwargs) ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/app/asu/build.py", line 308, in build report_error(job, "Impossible package selection") File "/app/asu/util.py", line 321, in report_error raise RuntimeError(msg) RuntimeError: Impossible package selection `

The router model target is the ASUS RT-AX59U.

a-gave, 2025-06-11:
Hi thanks for reporting, I just tried to reproduce and i got the same error

from here https://sysupgrade.antennine.org/api/v1/build/3c3e00b91bcf95b6271cfda007707e7e4979352f51a0f3f21a041b73de4584bc `cannot find dependency libuci20130104 for ubus-lime-utils` and `ubus-lime-utils found, but incompatible with the architectures configured`

pkg_hash_check_unresolved: cannot find dependency libuci20130104 for ubus-lime-utils\n * pkg_hash_fetch_best_installation_candidate: Packages for ubus-lime-utils found, but incompatible with the architectures configured\n * satisfy_dependencies_for: Cannot satisfy the following dependencies for lime-app:\n * \tlibuci20130104\n * opkg_install_cmd: Cannot install package lime-app.\nmake[2]: *** [Makefile:234: package_install] Error 255\nmake[1]: *** [Makefile:171: _call_manifest] Error 2\nmake: *** [Makefile:349: manifest] Error 2\n"

I try to investigate a bit, and reproduce with a plain imagebuilder to understand better what is the issue, and report here!

a-gave, 2025-06-11:
(Sorry the previous link is already expired since the build contains `defaults`)

Found! Packages at https://feed.libremesh.org/master are compiled against openwrt-23.05.5 * that contains the wrong libuci20130104

There is an active related pull request #1139 intended to build packages specifically for each architectures and that split the feeds for each openwrt branch (eg creating 
- https://feed.libremesh.org/arch_packages/master/openwrt-main/mips_24kc
- https://feed.libremesh.org/arch_packages/master/openwrt-24.10/mips_24kc
- https://feed.libremesh.org/arch_packages/master/openwrt-23.05/mips_24kc
)

But this issue is new, and the pull request #1139 should not be merged as it doesn't do yet the same splitting by `openwrt-branch` also for packages compiled for x86_64 (the main feed with packages that have`PKGARCH:=all`) and merging would override https://feed.libremesh.org/master thus allowing to build with a firmware with lime-app with the imagebuilder for openwrt-24.10.1 but loosing the possibility to build it for openwrt-23.05.5.

* https://mirror-03.infra.openwrt.org/releases/23.05.5/targets/ath79/generic/openwrt-23.05.5-ath79-generic.manifest
* https://downloads.openwrt.org/releases/24.10.1/targets/ath79/generic/openwrt-24.10.1-ath79-generic.manifest

closed 2025-07-02 by a-gave