libremesh/lime-packages #1197: Comparing LibreMesh

ilario, 2025-06-29:
I got some lime-reports to compare between LibreMesh on OpenWrt 24.10 and on OpenWrt 23.05. In the first case, I used the latest LibreMesh code (and including also the code from #1139 and #1164), while in the second I used the LibreMesh 2024.1 release. If we see anything dubious, I can compile the latest LibreMesh code on top of OpenWrt 23.05 also...

The employed routers are the same employed in #1192: YouHua WR1200JS (DSA), TP-Link WDR3600 (swconfig), TP-Link WR841N v13 (single band, swconfig), PlasmaCloud PA1200 (DSA, without changing the default ath10k driver selection which is ath10k-ct).

Overall, the issues I observed are: 

* #1192 so that the gateway via cable works intermittently.
* TP-Link WR841N v13 (64 MB RAM) has a high load average and does some boot looping. But I understand this is normal as 64 MB is too much for the default packages set and OpenWrt 24.10. Deselecting lime-app and pirania and avoiding wpad-mesh-* should help.
* #1191 shared-state-async dependency libstdcpp is too large to be installed on TP-Link WDR3600
* #1151 but I did not check again everything

The cabled network topology is the following:

                          TP-Link WR841N v13
                                      |
    laptop--YouHua WR1200JS--TP-Link WDR3600--PlasmaCloud PA1200--internet

As the routing protocols on the cable are active by default only on swconfig routers, via cable I would expect only the two TP-Link routers to see each other.

Here goes the output of lime-report on each router when using a firmware compiled on OpenWrt 24.10:

* TP-Link WR841N v13 [TP-Link_WR841N_v13-LibreMesh_master-OpenWrt_24.10-report.txt](https://github.com/user-attachments/files/20967732/TP-Link_WR841N_v13-LibreMesh_master-OpenWrt_24.10-report.txt)
* YouHua WR1200JS [YouHua_WR1200JS-LibreMesh_master-OpenWrt_24.10-report.txt](https://github.com/user-attachments/files/20967734/YouHua_WR1200JS-LibreMesh_master-OpenWrt_24.10-report.txt)
* TP-Link WDR3600 [TP-Link_WDR3600-LibreMesh_master-OpenWrt_24.10-report.txt](https://github.com/user-attachments/files/20967736/TP-Link_WDR3600-LibreMesh_master-OpenWrt_24.10-report.txt)
* PlasmaCloud PA1200 [PlasmaCloud_PA1200-LibreMesh_master-OpenWrt_24.10-report.txt](https://github.com/user-attachments/files/20967739/PlasmaCloud_PA1200-LibreMesh_master-OpenWrt_24.10-report.txt)

I tried to compile the same on OpenWrt 23.05 but got a `ERROR: tools/squashfs3-lzma failed to build.`, so I will post here those lime-reports as soon as I manage to compile it on another laptop.

ilario, 2025-07-06:
Still with OpenWrt 24.10:
 
> * [Default anygw route working intermittently via cable #1192](https://github.com/libremesh/lime-packages/issues/1192) so that the gateway via cable works intermittently.

In this setup, the result of the ping to the gateway via cable is:

```
--- 10.13.0.1 ping statistics ---
2857 packets transmitted, 2036 received, +6 duplicates, 28.7364% packet loss, time 2924195ms
rtt min/avg/max/mdev = 0.363/0.616/1.840/0.122 ms
```

Pinging 10.13.0.1 when associated to the ap_named WiFi interface of each router resulted like this:

Associated to the PlasmaCloud PA1200 named AP

ilario, 2025-07-06:
Still on OpenWrt 24.10:

> * [Default anygw route working intermittently via cable #1192](https://github.com/libremesh/lime-packages/issues/1192) so that the gateway via cable works intermittently.

In this setup, the result of the ping to the gateway via cable is:

```
--- 10.13.0.1 ping statistics ---
2857 packets transmitted, 2036 received, +6 duplicates, 28.7364% packet loss, time 2924195ms
rtt min/avg/max/mdev = 0.363/0.616/1.840/0.122 ms
```

---

Pinging 10.13.0.1 when associated to the ap_named interface of each router gave less-than-perfect results, here you can see them.

Associated to the PlasmaCloud PA1200 named AP: close to no packet loss, average one hundred ms.

```
--- 10.13.0.1 ping statistics ---
341 packets transmitted, 340 received, 0.293255% packet loss, time 340710ms
rtt min/avg/max/mdev = 2.556/101.466/2186.274/269.527 ms, pipe 3
--- 4.2.2.2 ping statistics ---
344 packets transmitted, 341 received, 0.872093% packet loss, time 343917ms
rtt min/avg/max/mdev = 29.845/140.103/2263.815/278.503 ms, pipe 3
```

Associated to the TP-Link WDR3600 named AP: low packet loss, average 0.7-1 s.

```
--- 10.13.0.1 ping statistics ---
473 packets transmitted, 473 received, 0% packet loss, time 475456ms
rtt min/avg/max/mdev = 1.245/668.858/7608.648/1092.744 ms, pipe 8
--- 4.2.2.2 ping statistics ---
448 packets transmitted, 434 received, 3.125% packet loss, time 450836ms
rtt min/avg/max/mdev = 29.823/1169.695/11086.026/1838.395 ms, pipe 11
```

Associated to the TP-Link WR841N v13 named AP: I got disassociated, likely due to the low RAM of this device. Sometimes the AP was not even visible. High packet loss, high average ping time.

```
--- 10.13.0.1 ping statistics ---
335 packets transmitted, 181 received, +51 errors, 45.9701% packet loss, time 340685ms
rtt min/avg/max/mdev = 125.274/9409.036/32968.540/9788.108 ms, pipe 33
--- 4.2.2.2 ping statistics ---
338 packets transmitted, 169 received, +48 errors, 50% packet loss, time 343902ms
rtt min/avg/max/mdev = 348.591/9709.688/33857.928/9701.469 ms, pipe 34
```

Associated to the YouHua WR1200JS named AP: high packet loss, high ping time.

```
--- 10.13.0.1 ping statistics ---
447 packets transmitted, 218 received, +29 errors, 51.2304% packet loss, time 454768ms
rtt min/avg/max/mdev = 91.357/4320.037/26839.877/5950.690 ms, pipe 27
--- 4.2.2.2 ping statistics ---
443 packets transmitted, 208 received, +27 errors, 53.0474% packet loss, time 451134ms
rtt min/avg/max/mdev = 225.719/4334.390/24983.997/5538.212 ms, pipe 25
```

ilario, 2025-07-06:
I forgot to specify: all the routers are in the same cardboard box, so that the wifi connection should be as good as it can be.

Still on OpenWrt 24.10:

> * [Default anygw route working intermittently via cable #1192](https://github.com/libremesh/lime-packages/issues/1192) so that the gateway via cable works intermittently.

In this setup, the result of the ping to the gateway via cable is:

```
--- 10.13.0.1 ping statistics ---
2857 packets transmitted, 2036 received, +6 duplicates, 28.7364% packet loss, time 2924195ms
rtt min/avg/max/mdev = 0.363/0.616/1.840/0.122 ms
```

---

Pinging 10.13.0.1 when associated to the ap_named interface of each router gave less-than-perfect results, here you can see them.

Associated to the PlasmaCloud PA1200 named AP: close to no packet loss, average one hundred ms.

```
--- 10.13.0.1 ping statistics ---
341 packets transmitted, 340 received, 0.293255% packet loss, time 340710ms
rtt min/avg/max/mdev = 2.556/101.466/2186.274/269.527 ms, pipe 3
--- 4.2.2.2 ping statistics ---
344 packets transmitted, 341 received, 0.872093% packet loss, time 343917ms
rtt min/avg/max/mdev = 29.845/140.103/2263.815/278.503 ms, pipe 3
```

Associated to the TP-Link WDR3600 named AP: low packet loss, average 0.7-1 s.

```
--- 10.13.0.1 ping statistics ---
473 packets transmitted, 473 received, 0% packet loss, time 475456ms
rtt min/avg/max/mdev = 1.245/668.858/7608.648/1092.744 ms, pipe 8
--- 4.2.2.2 ping statistics ---
448 packets transmitted, 434 received, 3.125% packet loss, time 450836ms
rtt min/avg/max/mdev = 29.823/1169.695/11086.026/1838.395 ms, pipe 11
```

Associated to the TP-Link WR841N v13 named AP: I got disassociated, likely due to the low RAM of this device. Sometimes the AP was not even visible. High packet loss, high average ping time.

```
--- 10.13.0.1 ping statistics ---
335 packets transmitted, 181 received, +51 errors, 45.9701% packet loss, time 340685ms
rtt min/avg/max/mdev = 125.274/9409.036/32968.540/9788.108 ms, pipe 33
--- 4.2.2.2 ping statistics ---
338 packets transmitted, 169 received, +48 errors, 50% packet loss, time 343902ms
rtt min/avg/max/mdev = 348.591/9709.688/33857.928/9701.469 ms, pipe 34
```

Associated to the YouHua WR1200JS named AP: high packet loss, high ping time.

```
--- 10.13.0.1 ping statistics ---
447 packets transmitted, 218 received, +29 errors, 51.2304% packet loss, time 454768ms
rtt min/avg/max/mdev = 91.357/4320.037/26839.877/5950.690 ms, pipe 27
--- 4.2.2.2 ping statistics ---
443 packets transmitted, 208 received, +27 errors, 53.0474% packet loss, time 451134ms
rtt min/avg/max/mdev = 225.719/4334.390/24983.997/5538.212 ms, pipe 25
```

a-gave, 2025-07-06:
A quick report:
I'm testing the solution proposed by pony in https://github.com/libremesh/lime-packages/issues/1121#issuecomment-2526310092
And it seems to work pretty well

create a new device bridge `br0` with all dsa user ports
add the bridge `br0` as member of `br-lan` and remove all dsa user ports from being direct members of `br-lan`
configure the 8021ad vlans for babel and batman on top of br0 (`br0_17` and `br0_29`)

I'm continuing these tests while preparing a pull request modifying /usr/lib/lua/lime/network.lua with these changes