libremesh/lime-packages #1206: default pirania handling do not support https request into mobile connected to client nodes

HarshilSoni849, 2025-07-15:
Hello there ,

I have installed lated v24.0 libremesh into my openwrt one router ,
default mesh network with internet sharing is working properly ,

then from libremesh page of gateway node into which internet is provided through ethernet , we configure and enabled the captive portal support and also created vouchers ,

when we connect AP of main gateway node , then it is opening the pirania captive portal as well as after applying voucher we can able to access internet into the mobile connected to AP of gateway.

but when we connect to AP of the client nodes connected to the gateway nodes , captive portal is opening but after applying voucher internet is not coming , only messaging in telegram and whatsapp is happaning but any website like youtube and facebook is not opening . 

so our query is ,
- is pirania handling the https request ?
- and what is solution of this issue which is explained above , we tested multiple time by flashing multiple images but issue is same.

ilario, 2025-07-15:
Thanks for the report!

The maintainer of Pirania is Altermundi, so you can additionally write them an email for bringing their attention on this issue:
https://github.com/libremesh/lime-packages/blob/12a078d0f9699479b7c12367a12e1777c8205781/packages/pirania/Makefile#L14

@luandro @henmohr @luandro @G10h4ck can you have a look at this?

HarshilSoni849, 2025-07-16:
Hello @ilario ,

Thank you for the information ,
But request you to please also look into this issue and let us know your suggestions and clarification based on your expertise.

luandro, 2025-07-17:
Hey @HarshilSoni849, thanks for your interest in Pirania and looking into it. We're well aware of all the issues, and there are several. The latest version is in the PR #1164, I suggest testing it out. But still, there are many functionalities lacking, and we're slowly working on them, as currently we lack any funding or any sort of support to do so. But we're actively looking into it.

luandro, 2025-07-29:
@HarshilSoni849 you can use this script: https://gist.github.com/luandro/de0ff7990ba8535f566e4c3e381dc4d3

1. Place it on the root of `lime-packages` folder
2. Make it runnable`chmod +x ./updatepkg`
3. Use it with:
```
./updatepkg.sh 10.13.0.1 pirania
```

Substituting with the correct IP of your router.

HarshilSoni849, 2025-07-30:
thank you for the script @luandro  ,

but i have one query that your code is in the  pull request https://github.com/libremesh/lime-packages/pull/1164 , and it seems like not merged in master,
so using your provided script , do we get the latest changes you provided into the pull request ?

ParthPrajapati-lab, 2025-09-04:
Hello @luandro,

I tested your latest pull request for the Pirania captive portal script and encountered the same issues mentioned earlier by @HarshilSoni849.

I made a few modifications to the script and was able to resolve some of the problems. Could you please review my updated script? There’s still one issue pending:

Issue: Internet accessible without applying voucher → Solved

Issue: DNS not resolving on nearest mesh node → Solved

Issue: Internet not working on mesh node AP after applying voucher → Pending

This is captive portal updated script:

#!/bin/sh
# requires nftables, liblucihttp0, liblucihttp-lua, uhttpd, uhttpd-mod-lua, uhttpd-mod-ubus

# Get the list of allowed MACs and networks
get_allowed_macs() {
    if [ -x /usr/bin/pirania_authorized_macs ]; then
        /usr/bin/pirania_authorized_macs || echo "Error: Failed to get authorized MACs" >&2
    else
        echo "Error: pirania_authorized_macs script not found" >&2
    fi
}

get_allow_list_ipv4() {
    # Get IPv4 allowlist from UCI
    uci get pirania.base_config.allowlist_ipv4 ||
        echo "10.0.0.0/8 172.16.0.0/12 192.168.0.0/16" # Default if not found
}

get_allow_list_ipv6() {
    # Get IPv6 allowlist from UCI
    uci get pirania.base_config.allowlist_ipv6 ||
        echo "fc00::/7 fe80::/64" # Default if not found
}

get_local_networks() {
    # Get local networks from network config
    uci show network | grep "ipaddr" | cut -d"'" -f2 ||
        echo "Error: Failed to get local networks" >&2
}

setup_nft_rules() {
    # Flush existing rules
    #nft flush ruleset
	
	if nft list tables inet | grep -q "pirania"; then
		nft delete table inet pirania
	fi

    # Create base tables and chains - keeping this close to the original working version
    nft add table inet pirania
    nft add chain inet pirania input { type filter hook input priority 0 \; }
    nft add chain inet pirania forward { type filter hook forward priority 0 \; }
    nft add chain inet pirania prerouting { type nat hook prerouting priority 0 \; }

    # Create sets for authorized MACs and allowed IPs
    nft add set inet pirania auth_macs { type ether_addr\; }
    nft add set inet pirania allow_ipv4 { type ipv4_addr \; flags interval \; }
    nft add set inet pirania allow_ipv6 { type ipv6_addr \; flags interval \; }

    # Allow established/related connections
    nft add rule inet pirania input ct state established,related accept
    nft add rule inet pirania forward ct state established,related accept
	
	# Allow the mesh router itself to bypass captive portal
	nft add rule inet pirania forward ip saddr 10.13.0.1 accept
	nft add rule inet pirania forward ip saddr 10.13.163.0 accept
[... 133 lines cut ...]
        ;;
    "update")
        logger -t "pirania" -p info "Updating captive portal rules"
        update_rules
        ;;
    "status")
        echo "Captive Portal Status:"
        echo "======================"
        echo "NFT Rules:"
        nft list table inet pirania 2>/dev/null || echo "Table not found"
        echo ""
        echo "Authorized MACs:"
        ubus call pirania get_authorized_macs
        echo ""
        echo "Services:"
        echo "  DNS Redirect (59053): $(netstat -lnu 2>/dev/null | grep -q ":59053" && echo "RUNNING" || echo "NOT RUNNING")"
        echo "  HTTP Redirect (59080): $(netstat -lnt 2>/dev/null | grep -q ":59080" && echo "RUNNING" || echo "NOT RUNNING")"
        ;;
    *)
        echo "Usage: $0 start|stop|update|status"
        exit 1
        ;;
esac

Thanks for your support!